SensorsIot/IOTstack #602: Cannot pull mosquitto

dmenne, 2022-09-22:
When I run the generated yml, I get the following error message:

```
 => ERROR [iotstack-mosquitto 3/7] RUN apk update && apk add --no-cache rsync tzdata                                     22.5s
------
 > [iotstack-mosquitto 3/7] RUN apk update && apk add --no-cache rsync tzdata:
#0 5.415 fetch http://dl-cdn.alpinelinux.org/alpine/v3.16/main/aarch64/APKINDEX.tar.gz
#0 10.42 ERROR: http://dl-cdn.alpinelinux.org/alpine/v3.16/main: temporary error (try again later)
#0 10.42 WARNING: Ignoring http://dl-cdn.alpinelinux.org/alpine/v3.16/main: No such file or directory
#0 10.42 fetch http://dl-cdn.alpinelinux.org/alpine/v3.16/community/aarch64/APKINDEX.tar.gz
#0 15.43 ERROR: http://dl-cdn.alpinelinux.org/alpine/v3.16/community: temporary error (try again later)
#0 15.43 WARNING: Ignoring http://dl-cdn.alpinelinux.org/alpine/v3.16/community: No such file or directory
#0 15.43 2 errors; 20 distinct packages available
------
failed to solve: executor failed running [/bin/sh -c apk update && apk add --no-cache rsync tzdata]: exit code: 2
```          

An explicit install works:

`docker pull eclipse-mosquitto:latest` # ok

A download of the requested file in the browsers also works:

`http://dl-cdn.alpinelinux.org/alpine/v3.16/main/aarch64/APKINDEX.tar.gz`

The error also occurs in a simplified compose.yaml with only

`docker compose -f compose.yaml build`

```
version: '3.6'

services:
  mosquitto:
    container_name: mosquitto
    build:
      context: ./.templates/mosquitto/.
      args:
      - MOSQUITTO_BASE=eclipse-mosquitto:latest
    restart: unless-stopped
    environment:
    - TZ=Etc/UTC
    ports:
    - "1883:1883"
    volumes:
    - ./volumes/mosquitto/config:/mosquitto/config
    - ./volumes/mosquitto/data:/mosquitto/data
    - ./volumes/mosquitto/log:/mosquitto/log
    - ./volumes/mosquitto/pwfile:/mosquitto/pwfile

```

Paraphraser, 2022-09-22:
My guess is you are running Buster without the `libseccomp2` patch. For some background see:

* [Issue 479](https://github.com/SensorsIot/IOTstack/issues/479)
* [Issue 401](https://github.com/SensorsIot/IOTstack/issues/401)

Assuming this diagnosis is correct, applying [the patch](https://sensorsiot.github.io/IOTstack/Basic_setup/#patch-2-update-libseccomp2) should fix it.

dmenne, 2022-09-22:
This is bullseye. Against the recommendation not to use it, I will think along these lines. I have already installed IOStack successfully some dozens of times with this image before. So it's probably caused by some recent update patch.

Linux raspberry 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64

```
lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
```

Paraphraser, 2022-09-22:
Test system where stack was not running (I've been using it for something else). Brought up the existing stack. Rebuilt Mosquitto. New image built and came up OK.

[image]

[image]

Paraphraser, 2022-09-22:
Not that it will likely make a difference but the REBUILD alias expands to:

```
docker-compose build --no-cache --pull mosquitto
```

Paraphraser, 2022-09-22:
The other thing that has **sometimes** caused build failures is DNS problems. Perhaps try some curls from that system to the URLs mentioned in the original output.

Paraphraser, 2022-09-22:
See [this](https://github.com/SensorsIot/IOTstack/issues/479#issuecomment-1016154791), then a [few posts later](https://github.com/SensorsIot/IOTstack/issues/479#issuecomment-1186540933) with tips on changing the DNS options.

dmenne, 2022-09-22:
Works now "suddenly", but sorry I cannot tell the world which of the many action I took really help - possibly some mystical cache expiry. I also cannot rule out that it has to do with http/https redirection, as intermediately curl worked with https and failed with http.

I also used https://www.schuetzler.net/blog/disable-pihole-for-some-devices/, but reverting it did not give a failure.

closed 2022-09-22 by dmenne