SensorsIot/IOTstack #603: Request for Apache Pulsar

tspannhw, 2022-09-23:
Please add Apache Pulsar to the mix.   Client (Python, Java, Golang, NodeJS) or server.

Paraphraser, 2022-09-23:
First off, I want to make it clear that I'm not in charge of anything. I'm just Joe User giving you some feedback which you're free to accept/reject.

<hr>

If you look on Dockerhub you'll see that [apachepulsar/pulsar](https://hub.docker.com/r/apachepulsar/pulsar/tags) and [apachepulsar/pulsar-all](https://hub.docker.com/r/apachepulsar/pulsar-all/tags) are only built for amd64. IOTstack is aimed at Raspberry Pi so it needs ARM.

There does seem to be an [open issue 12944](https://github.com/apache/pulsar/issues/12944) asking for this feature. It was opened November 2021 and has quite a few thumbs-ups. Having read to the end (which I suggest you do too), it seems they might be on the cusp of getting somewhere.

One of the posts along the way points to [kezhenxu94/pulsar](https://hub.docker.com/r/kezhenxu94/pulsar/tags). While it has a reasonable number of pulls, it hasn't been updated in 9 months, whereas the "official" image has been through quite a few evolutions in that time. I don't know how to interpret that.

The thing that gave me a case of instant hives was when I looked at the example [`docker-compose.yml`](https://github.com/apache/pulsar/blob/master/docker-compose/kitchen-sink/docker-compose.yml). My immediate question is "does this thing **really** need 15 separate containers?" If yes, maybe that explains why they haven't really focused on an ARM variant - it might be biting off a bit more than a Pi can chew. If no, where's the **simple** example? A lifetime in IT has taught me to be very suspicious when people go for complex out-of-the-box rather than starting simple and working up.

With all that in mind, can I suggest that you:

1. Leave this issue open.
2. Place a watch on issue 12944 and check the Dockerhub repo from time to time.
3. When an ARM image appears:

	* find or develop a **simple** (as in single-container) docker-compose service definition. If you need help adopting IOTstack conventions I'm happy to do that.
	* give it a whirl on your own system and write some how-to documentation so other people can get started.
	* either prepare a Pull Request or post your service definition here.

derskythe, 2023-04-07:
> Apache Pulsar

But why this product but not RabbitMQ?