SensorsIot/IOTstack #606: Backup script causing VLAN issue

yavinmaster, 2022-09-25:
I've been trying to get to a bottom of a problem that has plagued me for some time.  I have my IOT devices on a separate VLAN with a firewall rule to let me access them from the main LAN.  Every so often (once a month or so?) I can't access the Raspberry PI from the LAN, but if I put myself on the IOT VLAN then I can.  I can access other devices from the LAN that are on the VLAN so it doesn't look like a firewall issue.  It tends to "fix itself" the next day.
As my IOTstack install was quite old and I'd learned quite a bit since the first install, I recently did a clean install on the latest PI OS thinking that would be the end of my network issues.  Today however the access to VLAN was again lost.  After the usually log checks, reboots of the PI, firewall and network switch, I finally had a thought that the backup script runs at 2am every day.  I therefore ran it manually and my network access came back!!!!

Any idea what in the script is causing this issue - especially when a reboot of the PI doesn't fix it, but rerunning the script does?

I've checked the backup log from 2am this morning and the one that I ran this afternoon and there isn't anything that jumps out in the way of errors.

I won't be able to confirm this is definitely the issue until it happens again at some point, but it is the first time I've done something which appears to have immediately fixed the problem so the backup script appears to be the key.

Paraphraser, 2022-09-26:
Hi,

You haven't really given me enough to go on. Specifically, it isn't clear how your Raspberry Pi is connected. Is it:

1. The Pi plus the "other" IOT devices **only** attach to the VLAN;
2. The IOT devices **only** attach to the VLAN but the Pi attaches to **both** the main LAN and the VLAN; or
3. Something else?

It also isn't clear which device is implementing your firewall rules. Is that the Pi or something else?

A diagram of how you have things set up would really help.

<hr>

In the meantime, I'm going to make some (hopefully) educated guesses about what might be going on; at least to the extent of pointing to what might explain the observed behaviour of the problem resolving itself when you run the backup.

I'm assuming you mean the backup script which is supplied with IOTstack, and not my [IOTstackBackup](https://github.com/Paraphraser/IOTstackBackup) scripts.

One of the significant differences between the *supplied* backup script and IOTstackBackup is the *supplied* script takes your IOTstack down while the backup is running, whereas my IOTstackBackup scripts don't need to do that (one of the reasons why I wrote IOTstackBackup in the first place).

With that in mind, assume my stack is running:

```
$ DPS
NAMES          CREATED        STATUS
zigbee2mqtt    44 hours ago   Up 44 hours
wireguard      2 days ago     Up 2 days
pihole         2 days ago     Up 2 days (healthy)
portainer-ce   2 days ago     Up 2 days
mosquitto      3 days ago     Up 26 hours (healthy)
nodered        3 days ago     Up 3 days (healthy)
grafana        3 days ago     Up 3 days (healthy)
influxdb       3 days ago     Up 3 days (healthy)
```

How many lines of net filter rules are in place while the stack is running?

```
$ sudo nft list ruleset | wc -l
92
```

Let's simulate the effect of a backup with the "supplied script" by bouncing the stack and checking the filter tables as we go along:

```
$ DOWN
[+] Running 9/9
 ⠿ Container influxdb        Removed                                                                                   1.0s
 ⠿ Container zigbee2mqtt     Removed                                                                                   1.2s
 ⠿ Container portainer-ce    Removed                                                                                   0.9s
 ⠿ Container grafana         Removed                                                                                   1.1s
 ⠿ Container nodered         Removed                                                                                   1.0s
 ⠿ Container wireguard       Removed                                                                                  10.6s
 ⠿ Container mosquitto       Removed                                                                                   0.4s
 ⠿ Container pihole          Removed                                                                                   4.6s
 ⠿ Network iotstack_default  Removed                                                                                   0.2s

$ sudo nft list ruleset | wc -l
48

$ UP
[+] Running 9/9
 ⠿ Network iotstack_default  Created                                                                                   0.1s
 ⠿ Container pihole          Started                                                                                   2.9s
 ⠿ Container portainer-ce    Started                                                                                   3.3s
 ⠿ Container mosquitto       Started                                                                                   2.7s
 ⠿ Container nodered         Started                                                                                   2.3s
 ⠿ Container grafana         Started                                                                                   2.6s
 ⠿ Container influxdb        Started                                                                                   2.5s
 ⠿ Container wireguard       Started                                                                                   5.5s
[... 37 lines cut ...]
      driver: default
      config:
        - subnet: 172.30.0.0/22

  nextcloud:
    driver: bridge
    internal: true
    ipam:
      driver: default
      config:
        - subnet: 172.30.4.0/22

services:
…
```

That is, I put the `networks` section near the top (which makes it easier to concatenate new service definitions onto the end) and I have predictable subnets.

Predictable subnets mean I never run into Docker doing a random allocation that collides with something else I'm doing. My rule is 10/8 for ZeroTier, 172.16/12 for Docker, 192.168/16 for me. Having a known subnet for `iotstack_default` is also somewhere between *useful* and *essential* if you want to run BIND9 in a container (which I do).

> Truth to tell, I don't run NextCloud so I just comment-out the lines for the second subnet.

If your compose file is really ancient, I'd also advise a bit of compare/contrast between your active service definitions and those in the `.templates` folder. We keep improving things (eg Mosquitto and Node-RED now have better mechanisms for passing build-time arguments) but the menu lacks the smarts to detect that and at least say "hey - have you considered..."

Anyway, hope this helps.

yavinmaster, 2022-10-09:
OK.  It happened again today and I did some tests before and after taking the stack down and bringing it back up again.  Doing this did result in the network working again.  I didn't do the tcpdump but the netstat -rn showed a difference between the working and non-working system:

Not working:

`Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         192.168.30.1    0.0.0.0         UG        0 0          0 eth0
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth9551033
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth31436bd
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth949ff3b
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 vethd91d085
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth9ee4d2d
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth8977e93
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth209ad03
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth8e14561
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth6f6c5f0
172.17.0.0      0.0.0.0         255.255.0.0     U         0 0          0 docker0
192.168.0.0     0.0.0.0         255.255.240.0   U         0 0          0 br-cfc431c5fcd8
192.168.30.0    0.0.0.0         255.255.255.0   U         0 0          0 eth0`

Working:
`Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         192.168.30.1    0.0.0.0         UG        0 0          0 eth0
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 vetha16d81c
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth3328f10
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth3ea28dc
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 vethcfc164a
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth29668e6
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth83922ce
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth66ec22e
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth7f6e098
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 veth7dc8fe5
172.17.0.0      0.0.0.0         255.255.0.0     U         0 0          0 docker0
192.168.30.0    0.0.0.0         255.255.255.0   U         0 0          0 eth0
192.168.32.0    0.0.0.0         255.255.240.0   U         0 0          0 br-2f02f89b13e8`

I'm out of my depth on what is going on here but the bridge at the end appears to be different.  My LAN is on 192.168.1.x, the VLAN is 192.168.30.x.  I have no idea what 192.168.32.0 is but it is working when it is present.

Any thoughts would be much appreciated.

Paraphraser, 2022-10-09:
I think I know what might be going on but I won't know for certain until I can get some more information.

Please provide:

1. Your `docker-compose.yml`, suitably edited to obfuscate anything sensitive like passwords. If you use any override files, include those too, please.
2. The output from running the following commands:

	```
	$ uname -a
	$ grep "VERSION=" /etc/os-release
	$ grep -e "allowinterfaces" -e "denyinterfaces" /etc/dhcpcd.conf
	$ docker network ls
	$ docker ps --format "table {{.Names}}\t{{.RunningFor}}\t{{.Status}}"
	```

Please wrap your `docker-compose.yml` and the console output from running the commands inside triple-backtick lines (otherwise known as "code fences"):

\`\`\`
output here
\`\`\`

Code-fences use monospaced font and respect end-of-lines so the original layout is preserved and everything is much easier to read.

yavinmaster, 2022-10-09:
Thanks for your support with this @Paraphraser .  Requested information is below:

```
docker-compose.yml:

version: '3.6'
services:
  home_assistant:
    container_name: home_assistant
    image: ghcr.io/home-assistant/home-assistant:stable
  #image: ghcr.io/home-assistant/raspberrypi3-homeassistant:stable
  #image: ghcr.io/home-assistant/raspberrypi4-homeassistant:stable
    restart: unless-stopped
    network_mode: host
    volumes:
    - /etc/localtime:/etc/localtime:ro
    - ./volumes/home_assistant:/config
    - /var/run/dbus/system_bus_socket:/var/run/dbus/system_bus_socket
    devices:
    - "/dev/ttyAMA0:/dev/ttyAMA0"
    - "/dev/vcio:/dev/vcio"
    - "/dev/gpiomem:/dev/gpiomem"
    privileged: true
  mosquitto:
    container_name: mosquitto
    build:
      context: ./.templates/mosquitto/.
      args:
      - MOSQUITTO_BASE=eclipse-mosquitto:latest
    restart: unless-stopped
    environment:
    - TZ=Etc/UTC
    ports:
    - "1883:1883"
    volumes:
    - ./volumes/mosquitto/config:/mosquitto/config
    - ./volumes/mosquitto/data:/mosquitto/data
    - ./volumes/mosquitto/log:/mosquitto/log
    - ./volumes/mosquitto/pwfile:/mosquitto/pwfile

  nodered:
    container_name: nodered
    build: ./services/nodered/.
    restart: unless-stopped
    user: "0"
    environment:
    - TZ=Etc/UTC
    ports:
    - "1880:1880"
    volumes:
    - ./volumes/nodered/data:/data
    - ./volumes/nodered/ssh:/root/.ssh
    - /var/run/docker.sock:/var/run/docker.sock
    - /var/run/dbus/system_bus_socket:/var/run/dbus/system_bus_socket
    devices:
    - "/dev/ttyAMA0:/dev/ttyAMA0"
    - "/dev/vcio:/dev/vcio"
    - "/dev/gpiomem:/dev/gpiomem"

  portainer-ce:
    container_name: portainer-ce
    image: portainer/portainer-ce
    restart: unless-stopped
    ports:
    - "8000:8000"
    - "9000:9000"
    volumes:
    - /var/run/docker.sock:/var/run/docker.sock
    - ./volumes/portainer-ce/data:/data

[... 149 lines cut ...]

```
docker network ls
NETWORK ID     NAME               DRIVER    SCOPE
e597beb4866d   bridge             bridge    local
e5ca99c1dfaa   host               host      local
2f02f89b13e8   iotstack_default   bridge    local
ad433d1b4d2a   none               null      local
```

```
docker ps --format "table {{.Names}}\t{{.RunningFor}}\t{{.Status}}"
NAMES                   CREATED       STATUS
zigbee2mqtt             5 hours ago   Up 5 hours
eufy-security           5 hours ago   Up 5 hours
mosquitto               5 hours ago   Up 5 hours (healthy)
rtsp-simple-server      5 hours ago   Up 5 hours
portainer-ce            5 hours ago   Up 5 hours
home_assistant          5 hours ago   Up 5 hours
zigbee2mqtt_assistant   5 hours ago   Up 5 hours
unifi-controller        5 hours ago   Up 5 hours
nodered                 5 hours ago   Up 5 hours (healthy)
duplicati               5 hours ago   Up 5 hours

```

Paraphraser, 2022-10-10:
OK. I think I can explain the problem and tell you how to fix it.

## tl;dr

Add the following line to `/etc/resolv.conf`:

```
allowinterfaces eth*,wlan* 
```

Then take down your stack and reboot your machine. Taking the stack down ensures that docker cleans up the networks properly so please don't skip that bit.

## The details

What follows is me responding to your "I'm out of my depth on what is going on here". I'll try to flesh out the picture with my understanding of how it all hangs together.

I might be wrong about some of the details, of course. But I always hope that, if I say something wrong or dumb, people with more knowledge might read it and elect to share.
 
Here's the equivalent (relevant) output from my system:

```
$ head -22 IOTstack/docker-compose.yml
version: '3.6'

networks:

  default:
    driver: bridge
    ipam:
      driver: default
      config:
        - subnet: 172.30.0.0/22

# nextcloud:
#   driver: bridge
#   internal: true
#   ipam:
#     driver: default
#     config:
#       - subnet: 172.30.4.0/22

services:

  portainer-ce:
  …
```

```
$ grep -e "allowinterfaces" -e "denyinterfaces" /etc/dhcpcd.conf
allowinterfaces eth*,wlan*
```

```
$ docker network ls
NETWORK ID     NAME               DRIVER    SCOPE
530d50e0ece0   bridge             bridge    local
a99c0c6d43bd   host               host      local
1b18f152b579   iotstack_default   bridge    local
848b257e3b00   none               null      local
```

```
$ docker ps --format "table {{.Names}}\t{{.RunningFor}}\t{{.Status}}"
NAMES          CREATED        STATUS
wireguard      41 hours ago   Up 41 hours
influxdb       3 days ago     Up 3 days (healthy)
pihole         2 weeks ago    Up 2 weeks (healthy)
portainer-ce   2 weeks ago    Up 2 weeks
nodered        3 weeks ago    Up 9 days (healthy)
mosquitto      5 weeks ago    Up 4 weeks (healthy)
[... 149 lines cut ...]
	Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
	169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 eth1.2
	172.17.0.0      0.0.0.0         255.255.0.0     U         0 0          0 docker0
	172.30.0.0      0.0.0.0         255.255.252.0   U         0 0          0 br-a3b64c14a10d
	192.168.134.0   0.0.0.0         255.255.255.128 U         0 0          0 eth0.1
	192.168.134.128 0.0.0.0         255.255.255.128 U         0 0          0 eth0.2
	```
	
	The primary physical port `eth0` trunks two VLANs, `eth0.1` (for "house") and `eth0.2` (for "DMZ") to a managed switch. Routers really need static IP addresses so DHCP needs to stay out of the way.
	
	The secondary physical port `eth1	` will connect to the ISP. It needs to be tagged as VLAN ID 2 because that's an ISP requirement. When PPP comes up as `ppp0` it will obtain its IP address from the ISP. Again, DHCP needs to stay away from all of that.
	
	> At the moment, the Ethernet cable to the `eth1` port is connected to a hub so the interface will come up. A DHCP server is reachable via that hub but neither `eth1` nor `eth1.2` is asking for DHCP and neither has a static assignment. Getting a link-local IP in 169.254/16 proves DHCP has been told to keep mum.

	Overall:
	
	```
	allowinterfaces wlan*
	```
	
	So, if I enable WiFi then it will get a dynamic assignment whereas everything else is left alone.
	
Will any of this change when Network Manager becomes the go-to solution for Raspberry Pi's (instead of dhcpcd)? No idea.

Hope this helps.

yavinmaster, 2022-10-10:
WOW! What a reply.  Thank you so much @Paraphraser.  I will complete your suggestions at the first opportunity.  It will however likely take several re-reads to fully understand all of the background information you have provided but I will definitely learn from it and hopefully others will too.