SensorsIot/IOTstack #632: Support for ring-mqtt

basil-dsouza, 2022-12-23:
Has anyone considered adding ring-mqtt support into IOTStack?  Based on my limited knowledge of IOTStack, it seems like it should be possible to integrate this: https://github.com/tsightler/ring-mqtt/wiki/Installation-(Docker)
```
  ring-mqtt:
    container_name: ring-mqtt
    restart: unless-stopped
    image: tsightler/ring-mqtt
    ports:
      - 8554:8554                      # Enable RTSP port for external media player access
    volumes:
      - <host_path>:/data           # Mapping of local folder to provide persistent storage
    logging:                           #limit logs to 10m and 3 files
      options:
        max-size: 10m
        max-file: "3"

```

Am interested in hearing if anyone has already tried this? or if anyone is interested in trying it?
I know enough docker to add this locally to my docker-compose file and try to get it working, but do not know enough IOTStack to integrate this well into the menu system.

Paraphraser, 2022-12-24:
I've submitted pull requests to implement this suggestion:

* #633 for master branch ("new menu")
* #634 for old-menu branch ("old menu")
* #635 for experimental branch ("new new menu")

I've no idea how long it will take before the PRs are accepted/rejected.

If you want to try it out in the meantime, follow [these instructions](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d#-testing-pending-pull-requests-) to apply #633 to your local copy of IOTstack. That assumes you are running on master branch ("new menu").

I've also added some documentation to the IOTstack Wiki which explains a couple of changes (setting time zone and the MQTT broker) that are either recommended or necessary when running ring-mqtt in a container. As a convention, we only add/update documentation on master branch because that is what drives the Wiki. In the interim, you can see the documentation [here](https://github.com/SensorsIot/IOTstack/blob/1fbffb5b4a3a26beab756e1157e6c97f8a734b30/docs/Containers/Ring-MQTT.md).

basil-dsouza, 2022-12-26:
Wow that was a really fast response and solution!

I had a look through the PRs, and I notice that for 633 (the PR to master) you have made a change to only the service.yml (ignoring the md file for a sec) and not any menu files. I dont know much about IOT Stack, but will that mean the menu picks it up automatically?

And reading through the MD file, the instructions are really helpful, I spent several hours trying to figure out some of those things! The part that I didnt find in there was on how to generate the refresh token. 
This was something I still havent tried either, mainly cause I wasnt sure if there was any difference in running the command when the image is in a docker compose file as opposed to a stand alone docker run. And also if that then needs the generated token to be stored somewhere or not.

https://github.com/tsightler/ring-mqtt/wiki/Installation-(Docker)#acquire-the-refresh-token
`docker run -it --rm --mount type=bind,source=<host_path>,target=/data --entrypoint /app/ring-mqtt/init-ring-mqtt.js tsightler/ring-mqtt
`

Paraphraser, 2022-12-26:
> I had a look through the PRs, and I notice that for 633 (the PR to master) you have made a change to only the service.yml (ignoring the md file for a sec) and not any menu files. I dont know much about IOT Stack, but will that mean the menu picks it up automatically?

Yes. To give you a more complete answer, IOTstack has three "menus":

* "new menu" (master branch) works by detecting the presence of a service definition in a templates folder. If you add something like:

	```
	~/IOTstack/.templates/fred/service.yml
	```
	
	then "fred" will show up automatically in the menu.
	
	If the setup for a particular container needs to go further, python scripts can be added to the template to do that but I try to avoid those like the proverbial plague.
	
	The downside of new-menu service definitions is they are "left shifted" by two spaces and **can't** be concatenated onto the compose file. If you want to bypass the menu and "do it yourself", you need to re-add those spaces. Something like:
	
	```
	$ cd ~IOTstack
	$ sed -e "s/^/  /" ./.templates/fred/service.yml >>docker-compose.yml
	```
 
* "old menu" (old-menu branch) has service definitions in its templates folder which **can** be concatenated directly, as in:

	```
	$ cd ~IOTstack
	$ cat ./.templates/fred/service.yml >>docker-compose.yml
	```

	However, the old menu doesn't auto-discover and needs two extra lines of code to be added to the menu script per service definition.
		
* "experimental" (experimental branch) is something I don't understand. I believe the basic idea is that the menu itself should run in a container. I tried to get it to work a couple of times but my system always stalled. I'm not sure of the exact status of this menu (whether it's in a holding pattern, is an idea now abandoned, etc). When I submit PRs, I "hope for the best" by trying to keep the experimental templates in parallel with those of new menu so as to minimise the overhead if development resumes.

Anyway, I hope that answers your question and helps you to see what's going on.

> And reading through the MD file, the instructions are really helpful, I spent several hours trying to figure out some of those things! The part that I didnt find in there was on how to generate the refresh token.

Full disclosure: I don't have any "ring" devices and I don't understand anything about the "ring" ecosystem or environment. For me, this is all seat-of-the-pants.

When I constructed the pull requests, I:

1. Did a skim-read of the docs you pointed me at;
2. Re-jigged a couple of things in your original service definition to make it IOTstack-friendly;
3. Spun-up the container and nosed around.

My nosing around led me to discover the issue with `localhost:1883` so I made sure the "obvious solution" (editing the config file and restarting the container) would work and persist properly, and documented it.

Having done that, I fleshed-out the old-menu and experimental branches, wrote the doc, and turned it all into pull requests. 

That was as far as I went. I don't recall focusing on the specific issue of the token in my skim-read but if I noticed it, I probably looked at the `docker run` and assumed it would be taken care of correctly in a `docker-compose` scenario. And that assumption *was* well-founded … but I missed a step. My bad.

I won't give you a blow by blow but a bit more nosing around led me to this in the log:

```
2022-12-26T22:33:09.254Z ring-mqtt No refresh token was found in the state file, use the Web UI at http://<host_ip_address>:55123/ to generate a token.
2022-12-26T22:33:09.254Z ring-mqtt Succesfully started the token generator web UI
```

I did see those lines before but did not appreciate their significance. What they are telling us is:

1. The container also needs to map port 55123; and
2. The doc needs to include the instruction to fire-up the web page.

I've just amended the PRs accordingly. I can see the UI but, not having an account, I can't really test the process end-to-end with 2FA etc. I'll leave that to you. Please let me know if it works.

Paraphraser, 2022-12-26:
Doco with extra material is [here](https://github.com/SensorsIot/IOTstack/blob/50493a7b2c065343d6a00015905bf3a1a0134106/docs/Containers/Ring-MQTT.md)

PR #633 (2023-01-05): 2022-12-24 ring-mqtt - master branch - PR 1 of 3
Adds ring-mqtt service definition.

Adds documentation to Node-RED wiki page (master branch).

Fixes #632
commit 1fbffb5b (2023-01-05): 2022-12-24 ring-mqtt - master branch - PR 1 of 3

Adds ring-mqtt service definition.

Adds documentation to Node-RED wiki page (master branch).

Fixes #632
closed 2023-01-05 by Slyke