SensorsIot/IOTstack #643: Error getting docker version.

steam228, 2023-01-11:
IOT  stack suddenly stop working delivering the following error:

Error getting docker version. Error when connecting to docker daemon.
 Check that docker is running.                                              
                                                                          
Command: docker version -f "{{.Server.Version}}"                           
                                                                        
than it suggest upgrade mentioning it is outdated. On its turn the upgrade delivers a no need... current version ok message.

I'm far from knowledgeable about docker, so I need your help!

Thanks a lot!

Paraphraser, 2023-01-12:
It's usually more helpful if you include the actual complete message output. In issues, paste the output in between lines of triple back-ticks, as in:

\`\`\`<br>pasted text here<br>\`\`\`

That preserves the original formatting and makes it clear what's coming from the Pi and what's coming from you.

It will also help if you run a few commands like:

```
$ docker version -f "{{.Server.Version}}"
20.10.22
$ docker-compose version
Docker Compose version v2.14.1
```

Please do the above if what I'm about to say doesn't fix your problem.

"Error when connecting to docker daemon" probably means what it says so please read:

* [Maintaining docker + docker-compose](https://github.com/Paraphraser/PiBuilder/blob/master/reinstallation.md)

and do what it says. I'd recommend the "nuclear option". Please pay careful attention to the need to reboot. I've lost count of the people complaining "it doesn't work" when they either overlooked or ignored the instruction to reboot.

With luck, the result should be the output above with docker 20.10.22 and docker-compose 2.14.1. Thereafter, both should be upgraded by routine `apt update ; apt upgrade`.

If that *doesn't* fix your problem then perhaps try:

* [Menu maintenance](https://github.com/Paraphraser/PiBuilder/blob/master/reset_menu_enviroment.md)

You will already have cloned PiBuilder so you can start at step 4.

The background to this is in Issue #601.

If none of that helps then we'll have to dig deeper. Perhaps also describe your system (Pi 3/4, RAM, SD or SSD, whether it's Buster or Bullseye, 32- or 64-bit, the output from `uname -a`, when you last did an `sudo apt update ; sudo apt upgrade`, and anything else you can think of that might lead to some "ahah" moments).

steam228, 2023-01-12:
Hello, sorry for the poor description. I thought was a common error... and thank you very much for your help!

System Pi 4 SSD bullseye 64 bit

```
$ uname -a
Linux loureiro 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
```

last update today, and upgrade too.

I believe I already messed it all up.
will start again from scratch and be more careful next time.

Paraphraser, 2023-01-13:
Raspberry Pi OS Bullseye isn't too bad (in the sense of stuff getting wrecked beyond recovery is rare) so, if it were me, I might persist a bit longer.

Did you try the steps in "Maintaining docker + docker-compose"? What happened?

If that didn't work at all (ie you still couldn't run `docker version`) then I'd agree you're headed for a rebuild because something deeper in the OS is all-fouled-up.

<hr>

If you're actually going to start over then I'd recommend a couple of things. First, assuming your stack isn't working (no containers running), it should be safe to grab all of your IOTstack persistent store "as is". Something like this will get the job done:

```
$ cd
$ touch rebuild-backup-volumes.tar.gz
$ sudo tar -czf rebuild-backup-volumes.tar.gz -C ~/IOTstack/volumes .
```

You could use the same approach to grab `~/IOTstack/services`, along with copies of your compose file, override files (if you use them) and `.env` (if you use that).

Fairly obviously you'll need to get those backup files off the machine. I use `scp` for that kind of thing.

Then, I'd suggest going the [PiBuilder](https://github.com/Paraphraser/PiBuilder) route to rebuild, starting from a clean Raspbian. That'll get you to a rock solid base install, plus docker, docker-compose and other dependencies, plus IOTstack cloned. It's "menu ready" at that point but you'd want to hold off on running the menu.

You can start move your compose and override files etc back into place. For the archives like the above, something like:

```
$ mkdir ~/rebuild-volumes
$ sudo tar -x --same-owner -z -f rebuild-backup-volumes.tar.gz -C ~/rebuild-volumes
```

That brings back everything with the ownership and permissions intact. If there are no errors and everything looks OK, you can then move that into place:

```
$ sudo mv ~/rebuild-volumes ~/IOTstack/volumes
```

Repeat the trick with services if you saved that and then:

```
$ cd ~/IOTstack
$ docker-compose up -d
```

Dunno if any of this will help. Fingers crossed...