SensorsIot/IOTstack #651: Executing the backup script gives me a weird prompt

IhsenBouallegue, 2023-01-23:
When I try to execute the backup script, I get the following message
[image]

No matter what I type, it won't proceed. I don't think this is normal or expected behaviour.

Paraphraser, 2023-01-23:
Please read [what is `sudo`](https://sensorsiot.github.io/IOTstack/Basic_setup/What-is-sudo/).

> And please don't think "I already know what `sudo` is - I can skip that". It's a page in the IOTstack Wiki that you really need to read before you go on.

What you are trying to do is run the backup script as the root user. It doesn't expect that, which is why there's a problem.

Many (but not all) IOTstack scripts have a check at the start to detect this and they will exit with the message "this script should NOT be run using sudo". It would be better if **all** the scripts had this treatment but it's a work in progress.

If you read that link above carefully, you should come away with a better appreciation of when you should **not** (most of the time) and should (rarely) use `sudo` to run a script.

What bothers me most about your screen shot is the `root@raspberrypi` prompt. That suggests you've either enabled the root account and then logged-in as root, or have done something like a `sudo -s` to get a privileged shell.

And then I wonder whether you're in the habit of doing that a lot when working with IOTstack.

IOTstack is **intended** to be installed in the home directory of a non-privileged user (eg the default user "pi"). Its scripts are **intended** to be run without `sudo`. The `docker` and `docker-compose` commands are **intended** to be run without needing to use `sudo`.

The default user (eg "pi") is a member of the so-called "sudoers" group which lets the unprivileged user (pi) escalate to root privileges, command by command. Most scripts take advantage of that by using `sudo` to execute certain commands *when it is necessary to do so*. That's why the link I gave you above suggests using `grep` to search scripts for "sudo" because that tells you the script should **not** be invoked using `sudo`:

```
$ grep -c sudo ~/IOTstack/scripts/backup.sh 
3
``` 

Similarly, on a proper installation of IOTstack, the default user ("pi") will be a member of the `docker` group. That provides all the permissions needed to run both `docker` and `docker-compose`. If you try running those commands and you get errors suggesting the need for `sudo`, it's actually telling you that your installation is somehow mucked up.

A lot of things flow from improper use of `sudo`. If you clone IOTstack from GitHub using sudo, quite a few things that should be owned by "pi" will be owned by "root" and then nothing will work properly. You use `sudo` to overcome those problems and that just compounds the bigger problem - a whole cascade of issues.

Bottom line: if you are finding that you need `sudo` a lot and/or you spend a lot of time inside a privileged shell, something may be wrong with your installation. You might be better advised to start over. If you decide to do that, reply and I'll give you some more tips.

IhsenBouallegue, 2023-01-23:
Hey @Paraphraser Thanks for the detailed explanation. I am no linux expert, so please bear my ignorance. However, I do not use sudo with IOTstack. In fact, I barely used any command outside the menu.sh. This time I resorted to running the command manually because running the backup from menu.sh prints `permission denied` errors and is a bit bugged.
[image]
Running the command normally without sudo prints the same but without `backup completed` at the end.
[image]
And that is why I used sudo because is explicitly said `Permission denied`

Paraphraser, 2023-01-24:
Hi,

Let's rewind a bit. I'll try to explain what's going on.

In your very first screen shot, the system prompt is:

```
root@raspberrypi:/home/pi/IOTstack#
```

That says you are running as root. The most likely way to get into that state is to login as "pi" and then run these commands:

```
$ cd IOTstack
$ sudo -s
```

[image]

> You can also replace `sudo -s` with `sudo bash`. It's exactly the same. Both commands mean, "become root and start a shell".

The only difference between the last line of my screen shot and the first line of yours is the system name (your Pi is named "raspberrypi" while mine is named "sec-dev").

Also look at the last character of each prompt. In the first two lines you'll see "$" which means "running as a non-privileged user". In the third line you'll see "#" which means "running as root". So, both the `root@` and the `#` confirm you are running as root in your first screen shot.

Now, let's try an experiment. In the following the `$` at the start of the first three lines is intended to convey "run this command as an unprivileged user", while the `#` in the remaining lines means, "this command is being run by a privileged user". This structure where the expected prompt is shown is a common convention and is intended as a sanity check to avoid running commands with the wrong level of privilege.

Anyway, here we go:

```
$ cd
$ touch user-file.txt
$ sudo -s
# touch root-file.txt
# mkdir root-dir
# touch root-dir/root-file.txt
# exit
```

[image]

In words:

1. `cd` with no arguments changes to your home directory.
2. `touch` creates an empty file, owned by the current user (pi) and with default permissions (which we will see in a bit).
3. `sudo -s` starts a shell with root privileges. The prompt changes to `#`.
4. Because we are running as root, the file created by `touch` will be owned by root.
5. Create a directory. We are still running as root so this directory will also be owned by root.
6. Create a file inside that directory. We are still running as root so that file, too, will be owned by root.
7. The `exit` command (a synonym for <kbd>control</kbd>+<kbd>d</kbd>) drops out of the root shell and we're back to being "pi" and you'll see the prompt has gone back to `$`.

Now let's assess the situation:

```
$ ls -ld . root-dir
$ ls -l user-file.txt root-file.txt root-dir/root-file.txt
```

[image]

The `d` on the `-ld` means "show the information about the *directory* rather than its *contents*". In words:

1. The `.` directory (the current working directory which is also the user's home directory) is owned by "pi".
2. The `root-dir` directory is owned by "root".
3. The two `root-file.txt` files are owned by "root".
4. The `user-file.txt` is owned by "pi".

Now, while running as "pi", let's try deleting all of that:

```
[... 191 lines cut ...]
That's a very powerful and dangerous command if you get it wrong so **please** make sure you get it exactly right before you press return.

<hr>

Earlier, I said I don't use the `backup.sh` script. What I use instead is:

* [IOTstackBackup](https://github.com/Paraphraser/IOTstackBackup)

The primary advantage of IOTstackBackup is that you don't need to stop your stack just to run a backup. The only "gotchas" are if you run either PostgreSQL or Subversion containers - those are just ignored by IOTstackBackup.

Also, the scripts in IOTstackBackup will not actually let you make the mistake of running them as the `root` user. That doesn't mean they won't chuck up errors if some other inappropriate use of `sudo` has somehow mucked up the expected structures inside the IOTstack directory but they won't be the *cause* of it.

<hr>

Please don't feel too bad about all this. You are far from the first person to run into strife with `sudo` (me included). That's why that page in the Wiki was written in the first place. The need to use `sudo` from time to time is a well-known Unix weakness.

> In macOS, Apple has tried to hide a lot of the details by having the system ask for permission to escalate privileges when necessary, as opposed to expecting every user to "just know" when they have to use `sudo` but the underlying problem is still there.

I do recommend that you adopt the approach proposed in that Wiki page. Try commands without `sudo` first, and only use `sudo` to force the issue if the first attempt fails. It doesn't guarantee you'll never hit this kind of problem again, just reduces the chances a bit.

Anyway, I hope this all helps. You might even know a bit more about Unix than you ever wanted so. Sorry about that.

<hr>

If you're wondering, the main reason I write long-winded and detailed answers to issues is because everything saved on GitHub persists. The next time this kind of problem comes up, I can just point the person to this issue.

IhsenBouallegue, 2023-01-28:
@Paraphraser I am out of words. Thank you so much for every detail. Sorry if I took too long to reply, it took quite a while and a few reads to understand and execute.
I followed your instruction to the letter. Sadly, the problem persisted. But because I have a better understanding and not only a list of commands, I debugged a bit. I checked the ownership of my backup directory (in the new IOTstack), turns out, it is root for some reason (probably a fault of mine). I changed it like we changed the other files and surprise surprise that solved it. But that opened my eyes to how messed up my setup might be. I was going to do a fresh install anyway because I bought an SSD for IOTstack (hence the backup). I am done living in the fear of an SD card fail 😆 
But I am going to do it right this time and use the PiBuilder tool (which I discovered through your github!) That way, I make sure everything is set up the right way.

Again, thank you for the detailed explanation and the awesome tools you wrote ❤️. This made me love IOTstack more. And especially trust myself around it more 😛 

I am closing this issue as the original problem was solved 👍

closed 2023-01-28 by IhsenBouallegue

Paraphraser, 2023-01-28:
Actually, that one is pretty rare and possibly the result of a `sudo git clone`. But, I did actually think of that, which is why I said to move the old IOTstack out of the way and clone a fresh copy without `sudo`. If that fresh copy somehow got root ownership then, I agree, something is so hosed-up somewhere that it probably really is time for the rebuild you're planning.

By the way, you should be able to install IOTstackBackup on your existing system and use that to take a "before" snapshot. A PiBuilder run includes IOTstackBackup (and clones IOTstack) so, basically, you get to the end of the PiBuilder run, execute the `iotstack_restore` command, and then tell your stack to come up.

IhsenBouallegue, 2023-01-28:
@Paraphraser that's exactly what I did, and now it works perfectly and as expected 👍 Thank you again 🙏

6ffm70, 2024-12-06:
@Paraphraser I tried to fix an error I get when backing up from the menu.
```
Execute Backup:
mkdir: cannot create directory ‘./backups/logs’: Permission denied
mkdir: cannot create directory ‘./backups/backup’: Permission denied
mkdir: cannot create directory ‘./backups/rolling’: Permission denied
touch: cannot touch './backups/logs/backup_2024-12-06_2110.log': No such file or directory
./scripts/backup.sh: line 70: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 71: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 72: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 73: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 74: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 77: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 81: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 86: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 87: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 88: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 101: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 105: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
cp: cannot stat './.tmp/backup/backup_2024-12-06_2110.tar.gz': No such file or directory
cp: cannot stat './.tmp/backup/backup_2024-12-06_2110.tar.gz': No such file or directory
./scripts/backup.sh: line 118: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
stat: cannot stat './.tmp/backup/backup_2024-12-06_2110.tar.gz': No such file or directory
./scripts/backup.sh: line 122: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 123: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 125: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 126: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 127: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 129: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 130: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
Something went wrong backing up. The temporary backup file doesn't exist. No temporary files were removed
Files: 
  ./.tmp/backup-list_2024-12-06_2110.txt
./scripts/backup.sh: line 146: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 147: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
./scripts/backup.sh: line 148: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory
cat: ./backups/logs/backup_2024-12-06_2110.log: No such file or directory

Backup completed.
```
Looking at the permissions in the folder, `/backups` belongs to `root`:
```
pi@6o594:~/IOTstack.old $ ls -l
total 124
drwxr-xr-x 3 root root  4096 Apr 26  2022 backups
-rw-r--r-- 1 pi   pi    4111 Jan  3  2023 docker-compose.yml
-rw-r--r-- 1 pi   pi    4165 Jan  3  2023 docker-compose.yml.bak
drwxr-xr-x 7 pi   pi    4096 Jul 11 01:39 docs
drwxr-xr-x 2 pi   pi    4096 Apr 26  2022 duck
-rwxr-xr-x 1 pi   pi   15483 Dec  3 15:59 install.sh
-rw-r--r-- 1 pi   pi   35149 Apr 26  2022 LICENSE
-rwxr-xr-x 1 pi   pi   13488 Dec  5  2022 menu.sh
-rw-r--r-- 1 pi   pi    2826 Dec  5  2022 mkdocs.yml
-rw-r--r-- 1 pi   pi    2860 Jul 11 01:39 README.md
-rw-r--r-- 1 pi   pi      49 Jul 11 01:39 requirements-menu.txt
-rw-r--r-- 1 pi   pi     412 Jul 11 01:39 requirements-mkdocs.txt
drwxr-xr-x 5 pi   pi    4096 Jul 11 01:39 scripts
drwxr-xr-x 4 pi   pi    4096 Apr 26  2022 services
drwxr-xr-x 9 root root  4096 Jan  3  2023 volumes
```
Going through your advice to create a new folder and getting a fresh IOTstack did not fix this issue. I did not use `sudo` where no explicitly required. Do you have any hint for me where to look?

Paraphraser, 2024-12-07:
>  I did not use sudo where no explicitly required.

I hear you but, even so, the most-likely explanation remains that you did this:

```
$ cd ~/IOTstack
$ sudo ./scripts/backup.sh
```

Depending on how many commands you've executed in the meantime, the proof might still be in your history file. Try this:

```
$ history | grep "./scripts/backup.sh"
```

You might get lucky and get a hit with `sudo` in it.

> Looking at the permissions in the folder, /backups belongs to root:
> ...
> getting a fresh IOTstack did not fix this issue

Yes. But you listed `~/IOTstack.old`. You said the same thing happened on a clean clone. What about a listing of that "clean" `~/IOTstack` folder?

To try to save time, I am going to make these assumptions:

1. The folder `~/IOTstack.old` is still as shown in your listing and that you haven't interfered with its contents or tried to blend it with `~/IOTstack`.
2. The folder `~/IOTstack` is present but is in some indeterminate state.
3. No containers are running.

## if those assumptions hold...

First, be in your home directory:

```
$ cd
``` 

Next, rename the `~/IOTstack` folder to get it out of the way:

```
$ mv IOTstack IOTstack.clean
```

Rename the `.old` folder so it becomes the active folder:

```
$ mv IOTstack.old IOTstack
```

Repair the original folder:

```
$ cd ~/IOTstack
$ sudo chown -R $USER:$USER backups
$ [ -d backups/influxdb ] && sudo chown -R root:root backups/influxdb
```

In words:

1. Be in the right directory.
2. Change ownership of the entire backups folder to be the current user.
3. If the backup folder for InfluxDB exists, change its ownership to root.

## if those assumptions don't hold...

Alternatively, if your "clean clone" is now your working IOTstack and you don't want to lose it then you can just repair it with the same commands (no need to stop any containers):

```
$ cd ~/IOTstack
$ sudo chown -R $USER:$USER backups
[... 66 lines cut ...]
scripts/backup.sh-  fi
scripts/backup.sh-else
scripts/backup.sh-  USER=$(whoami)
scripts/backup.sh-fi
```

No hits on EUID in `restore.sh` and whatever that code in `backup.sh` is doing, it isn't checking for EUID and refusing to run if the script is launched using `sudo`. Thus, because they don't stop you from using `sudo`, these two scripts are good candidates for causing this problem.

In a typical system, "running a backup" is performed more frequently than "running a restore" so, statistically, that makes `sudo ./scripts/backup.sh` the more-likely candidate.

Make sense?

<hr>

Last point. Perhaps take a look at [IOTstackBackup](https://github.com/Paraphraser/IOTstackBackup). 

> If you constructed your system using [PiBuilder](https://github.com/Paraphraser/PiBuilder) then IOTstackBackup will already be installed, otherwise installation instructions are in its README.

The IOTstackBackup scripts contain that check for EUID so you can't make the mistake of running them with `sudo`. As well as creating `backups` if it doesn't exist, they also check that folder's ownership and fix it on the fly if it is wrong.

Unlike the `backup.sh` script which requires you to take down your stack for the duration of the backup, the `iotstack_backup` script works in parallel with an active stack so you don't have to interrupt your IoT services just to take a backup. In fact, for database containers like InfluxDB, the container *must* be running for the backup to work.

Once you have configured IOTstackBackup, you can drive it from `cron` and let it take care of the details of (a) running the backup, (b) getting the backup files transferred to your choice of another machine in your local network or a cloud service like Dropbox, and (c) pruning old backups.

Nothing involving computers or networking is ever 100% foolproof but I've tried to make IOTstackBackup as close to set-and-forget as is humanly possible.

6ffm70, 2024-12-07:
Thank you for the in-depth reply, @Paraphraser 

Before I try your fixes, I've had a look at the history:
[image]

I've always attempted the backup with the menu, and can't remember using sudo; however that installation is a bit older already:
[image]

IOTstack.old is still the original installation directory; as per your instructions dated Jan 24, 2023 above. The folders compare like so:
[image]

I'll report back when I've done with the clean install and changing backups permissons.

Paraphraser, 2024-12-07:
I think I've figured it out. It wasn't until I saw your third screen shot that it hit me between the eyes like an Aussie magpie slamming into a window while chasing a cricket. I can be so ***dense*** sometimes!

But, before I get to that, the `history` command is global - not relative to the folder you're working in. The first command was sufficient to disprove my original theory.

OK. New hypothesis: this problem is the result of a series of events. I'll take you through what *should* happen because that will help you (and anyone else reading this issue in future) understand the context.

There are [two standard ways](https://sensorsiot.github.io/IOTstack/Basic_setup/#new-installation) to install IOTstack:

1. the add-on method where you use `curl` to fetch `install.sh` and pipe it to `bash`; and
2. the PiBuilder method.

It is also possible to simply do:

```
$ git clone https://github.com/SensorsIot/IOTstack.git ~/IOTstack
```

but you won't find that method recommended anywhere. That's because a direct clone doesn't satisfy dependencies and **doesn't initialise the basic environment properly**. (emphasis contains the clue).

If you go back to my earlier reply, one of the hits on things in IOTstack that contain `mkdir` and `backups` was in `install.sh`. The line of code shown in the hit was:

```
mkdir -p "$IOTSTACK/backups" "$IOTSTACK/services"
```

PiBuilder has the [same line of code](https://github.com/Paraphraser/PiBuilder/blob/a8c2c00c428e621bca79531799f6cae8afd70ada/boot/scripts/03_setup.sh#L223).

The take-home message at this point in the story is that, if IOTstack is installed via either its installer script or PiBuilder, the result is:

1. A clean clone of the IOTstack repo;
2. The creation of `~/IOTstack/backups`; and
3. The creation of `~/IOTstack/services`,

all of which are owned by the current user.

The next thing the first-time user is likely to do is:

```
$ cd ~/IOTstack
$ ./menu.sh
$ docker-compose up -d
```

Let's assume that that menu run included the MING stack (Mosquitto, InfluxDB, Node-RED, Grafana).

> Any collection of containers will do, providing it includes InfluxDB 1.8 because it's the necessary villain in this story.

Below, I'm using `grep` to find all the volume mappings in those containers.

```
$ cd ~/IOTstack/.templates
$ grep ' - ./' mosquitto/service.yml influxdb/service.yml nodered/service.yml grafana/service.yml 
mosquitto/service.yml:    - ./volumes/mosquitto/config:/mosquitto/config
mosquitto/service.yml:    - ./volumes/mosquitto/data:/mosquitto/data
mosquitto/service.yml:    - ./volumes/mosquitto/log:/mosquitto/log
mosquitto/service.yml:    - ./volumes/mosquitto/pwfile:/mosquitto/pwfile
influxdb/service.yml:    - ./volumes/influxdb/data:/var/lib/influxdb
influxdb/service.yml:    - ./backups/influxdb/db:/var/lib/influxdb/backup
nodered/service.yml:    - ./volumes/nodered/data:/data
nodered/service.yml:    - ./volumes/nodered/ssh:/root/.ssh
grafana/service.yml:    - ./volumes/grafana/data:/var/lib/grafana
grafana/service.yml:    - ./volumes/grafana/log:/var/log/grafana
```

As `docker-compose` is reading `docker-compose.yml`, it processes `volumes:` clauses by interpreting the leading `.` as "the path to the folder containing `docker-compose.yml`. The first line in the above will generally expand to something like:

```
  - /home/pi/IOTstack/volumes/mosquitto/config:/mosquitto/config
```

[... 14 lines cut ...]
If the `backups` folder was created by either `install.sh` or `PiBuilder` then it will already exist, in which case only `influxdb` and `db` will be created with root ownership.

Conversely, if `~/IOTstack` was the result of a direct clone then `backups` won't exist so it, too, will get `root` ownership.

Providing this explanation of how `backups` can come to be owned by root makes sense in terms of what you remember of how your IOTstack folders came to be created then it's the second most logical explanation for "backups" being owned by root (inadvertently running the backup script via `sudo` being the prime candidate).

I knew all this. I just don't see this problem sufficiently often for the explanation to be top-of-mind. My bad. Sorry.

Although I have been known to just clone IOTstack from the repo and run it "as is" without pre-creating either "services" or "backups", I never run the supplied backup script so I'd never encounter this problem. Also, as noted yesterday, the IOTstackBackup scripts check for `backups` being owned by root and fix it on the fly.

For completeness, the reason why "services" gets the same treatment as "backups" is because, historically, the Mosquitto container had volumes paths into the services directory. If the services directory didn't exist when `docker-compose` came along then it, too, would wind up being owned by root and the menu would have trouble making anything work.

We were able to solve the problem with Mosquitto but moving InfluxDB's backup folder into `volumes` is a little more tricky because we don't necessarily control all the backup scripts that might depend on the backup folder being where it is now, so we live with this problem.

> Making `~/IOTstack/backups/influxdb` the backup location was actually one of Graham Garner's decision which has been inherited by this SensorsIot fork. 

I did "solve" it for InfluxDB 2.0 but I'd never recommend anyone run that container because it's (a) hard to use and (b) appears to be a technological dead-end.

[... cut ...]
And, seeing as I've been thinking about ways in which stuff can wind up with the wrong ownership, there's also a third, which is the result of an improper installation of Docker. One of the key steps is adding the current user to the `docker` group. That gives the current user permission to run docker commands without needing sudo. If the current user isn't in the group then that produces a natural inclination to use `sudo` to force the issue, and then you get a mess.

Lastly, my apologies for being slow on the uptake.