SensorsIot/IOTstack #660: No module named 'blessed'

NossieUK, 2023-01-29:
pi@Europa:~/IOTstack $ ./menu.sh
Checking for project update
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
Project is up to date
Python virtualenv found.
Python Version: 'Python 3.11.1'. Python and virtualenv is up to date.
Please enter sudo pasword if prompted
Command: docker version -f "{{.Server.Version}}"
Docker version 20.10.23+dfsg1 >= 18.2.0. Docker is good to go.
Project dependencies up to date

Existing installation detected.
Using existing python virtualenv for menu
Traceback (most recent call last):
  File "/home/pi/IOTstack/./scripts/menu_main.py", line 2, in <module>
    from blessed import Terminal
ModuleNotFoundError: No module named 'blessed'

Just wondering if anyone has came across this?  I tried to install blessed with pip3 install blessed and it says it's installed but still does this?

Paraphraser, 2023-01-29:
See if [this)](https://github.com/Paraphraser/PiBuilder/blob/master/reset_menu_enviroment.md) helps. See also #643.

NossieUK, 2023-01-29:
so I ran the instructions from pibuilder and logged back in running the menu

      creating build/temp.linux-aarch64-cpython-311
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/IOTstack/.virtualenv-menu/include -I/usr/include/python3.11 -c _ruamel_yaml.c -o build/temp.linux-aarch64-cpython-311/_ruamel_yaml.o
      _ruamel_yaml.c:198:12: fatal error: longintrepr.h: No such file or directory
        198 |   #include "longintrepr.h"
            |            ^~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> ruamel.yaml.clib

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
pi@Europa:~/IOTstack $                      

How weird :-/

Paraphraser, 2023-01-29:
Maybe some basic info like your hardware, `uname -a` output, whether you're running Buster or Bullseye, etc.

Perhaps also try [this](https://github.com/Paraphraser/PiBuilder/blob/master/reinstallation.md).

I really have no idea what's going on. I don't understand this Python venv thing (conceptually, yes, tech details, no) but it seems to be the root cause of many problems.

NossieUK, 2023-01-29:
I've been using bookworm for months without issue...

pi@Europa:~/IOTstack $ uname -a
Linux Europa 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

Paraphraser, 2023-01-29:
Just to confirm, I ran the reset script (my first suggestion) on my own system and it didn't chuck up the same error as you got so either (a) it was a transient problem, in which case trying again might help, or (b) it's something peculiar about your system which is causing it.

I don't ever recall seeing compiler **involvement** let alone include errors or mentions of ruamel.yaml.clib. Weird.

Paraphraser, 2023-01-29:
Shows you how little I know - "bookworm" - never heard of it until this moment. Googling finds hits in 2021. Must've been living under a rock. What hardware?

NossieUK, 2023-01-29:
8gb raspberry pi4

Paraphraser, 2023-01-29:
We're at the limits of my knowledge. I usually try to treat the symptoms. On your system it's moaning about trying to install ruamel.yaml.clib.

On my Bullseye system:

```
$ apt list ruamel.yaml.clib python3-ruamel.yaml.clib
Listing... Done
python3-ruamel.yaml.clib/stable 0.2.2-1+b2 arm64
python3-ruamel.yaml.clib/stable 0.2.2-1+b2 armhf
```

So, `ruamel.yaml.clib` isn't known but `python3-ruamel.yaml.clib` is, albeit not marked "installed".

I just installed `python3-ruamel.yaml.clib`, then re-ran the reset script, then let the menu rebuild its venv. That all worked so it obviously did no great harm to install that package.

Maybe see if making the library available will solve the problem on Bookworm.

Other than that, serious "dunno". Sorry.

NossieUK, 2023-01-29:
I wonder if it's related to it not being a stable build and they have broke something reacently - I used bookworm on another pi and installed powerline shell as part of that.  After a recent update, the update broke that shell too and I had to reinstall the app to fix it.

pi@Europa:~/IOTstack $ apt list ruamel.yaml.clib python3-ruamel.yaml.clib
Listing... Done
python3-ruamel.yaml.clib/testing 0.2.7-1+b1 arm64
python3-ruamel.yaml.clib/testing 0.2.7-1+b1 armhf

I use a heavily modified docker-compose file that I write manually, I used IOTstack just to start stop / update containers etc 

I'll go back to using the docker commands manually and see if the next set of updates or so fixes it again \o/  Thank you for having a look!

NossieUK, 2023-03-11:
Just to add - all my installs using bookworm on arm fail this - I recently installed linux mint and using the defaults this installed fine - bookworm is still int testing but I suggest you look into this.

INFO:  Running under normal user some features may not work.
>> DECODEING MEMORY <<
Fan Status ON Speed 10%
System Temperature 48°
Hysteresis set to 10°
Fan Speeds set to 10% 55% 100%
Fan Temps set to 55° 60° 65°
Fan Mode [ AUTO ] 
Fan Speed Override 0% 
Target Temperature 0° 
Daemon Status : Waiting for request
Maximum Temperature : 67°
Minimum Temperature : 42°
Daemon Warnings : 0
Daemon Errors : 0
Daemon Critical Errors : 0
Fan Status ON Speed 10%
       _,met$$$$$gg.          pi@Europa 
    ,g$$$$$$$$$$$$$$$P.       --------- 
  ,g$$P"        """Y$$.".     OS: Debian GNU/Linux bookworm/sid aarch64 
 ,$$P'              `$$$.     Host: Raspberry Pi 4 Model B Rev 1.5 
',$$P       ,ggs.     `$$b:   Kernel: 5.15.84-v8+ 
`d$$'     ,$P"'   .    $$$    Uptime: 3 hours, 10 mins 
 $$P      d$'     ,    $$P    Packages: 2533 (dpkg), 6 (flatpak), 9 (snap) 
 $$:      $$.   -    ,d$$'    Shell: bash 5.2.15 
 $$;      Y$b._   _,d$P'      Resolution: 1920x1080, 1024x600 
 Y$$.    `.`"Y$$$$P"'         Terminal: /dev/pts/1 
 `$$b      "-.__              CPU: Cortex-A72 (4) @ 2.000GHz 
  `Y$$                        Memory: 3893MiB / 7384MiB 
   `Y$$.
     `$$b.                                            
       `Y$$b.                                         
          `"Y$b._
              `"""

pi@Europa:~ $ cd IOTstack/
pi@Europa:~/IOTstack $ ./menu.sh
Checking for project update
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
Project is up to date
Python virtualenv found.
Python Version: 'Python 3.11.2'. Python and virtualenv is up to date.
Please enter sudo pasword if prompted
Command: docker version -f "{{.Server.Version}}"
Docker version 20.10.23+dfsg1 >= 18.2.0. Docker is good to go.
Project dependencies up to date

Existing installation detected.
Creating python virtualenv for menu...
Installing menu requirements into the virtualenv...
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [25 lines of output]
      sys.argv ['/tmp/pip-install-m_gi1u0s/ruamel-yaml-clib_3d75878b1f5b48adbdbe2e4455327ce3/setup.py', 'bdist_wheel', '-d', '/tmp/pip-wheel-kiazuuot']
      test compiling /tmp/tmp_ruamel_27qgtiny/test_ruamel_yaml.c -> test_ruamel_yaml OK
      /home/pi/IOTstack/.virtualenv-menu/lib/python3.11/site-packages/setuptools/dist.py:285: SetuptoolsDeprecationWarning: The namespace_packages parameter is deprecated, consider using implicit namespaces instead (PEP 420).
        warnings.warn(msg, SetuptoolsDeprecationWarning)
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-311
      creating build/lib.linux-aarch64-cpython-311/ruamel
      copying .ruamel/__init__.py -> build/lib.linux-aarch64-cpython-311/ruamel
      creating build/lib.linux-aarch64-cpython-311/ruamel/yaml
[... 35 lines cut ...]
      creating build/lib.linux-aarch64-cpython-311/ruamel/yaml
      copying .ruamel/yaml/__init__.py -> build/lib.linux-aarch64-cpython-311/ruamel/yaml
      creating build/lib.linux-aarch64-cpython-311/ruamel/yaml/clib
      copying ./__init__.py -> build/lib.linux-aarch64-cpython-311/ruamel/yaml/clib
      copying ./LICENSE -> build/lib.linux-aarch64-cpython-311/ruamel/yaml/clib
      running build_ext
      building '_ruamel_yaml' extension
      creating build/temp.linux-aarch64-cpython-311
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/IOTstack/.virtualenv-menu/include -I/usr/include/python3.11 -c _ruamel_yaml.c -o build/temp.linux-aarch64-cpython-311/_ruamel_yaml.o
      _ruamel_yaml.c:198:12: fatal error: longintrepr.h: No such file or directory
        198 |   #include "longintrepr.h"
            |            ^~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> ruamel.yaml.clib

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
pi@Europa:~/IOTstack $

Paraphraser, 2023-03-11:
Perhaps see if [this](https://github.com/Paraphraser/PiBuilder/blob/master/reset_menu_enviroment.md) helps.

NossieUK, 2023-03-12:
Sorry - just to clarify Paraphraser - this appears to be an issue with either a change or something to debian or python -   I'm going to double check but I'm pretty sure it will do this with a fresh bullseye install that has been updated to bookworm - I don't think it's an IOTstack issue,. more than it being a headsup of what's around the corner.

pi@Europa:~ $ git -C ~/PiBuilder pull origin master
From https://github.com/Paraphraser/PiBuilder
 * branch            master     -> FETCH_HEAD
Already up to date.
pi@Europa:~ $  ./PiBuilder/boot/scripts/helpers/reset_menu_enviroment.sh

Ensuring apt directories are up-to-date...
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Get:2 http://deb.debian.org/debian bookworm InRelease [178 kB]                 
Hit:3 http://security.debian.org/debian-security bookworm-security InRelease   
Hit:4 https://download.docker.com/linux/debian bullseye InRelease              
Get:5 http://deb.debian.org/debian bookworm-updates InRelease [49.6 kB]        
Hit:7 https://download.sublimetext.com apt/stable/ InRelease                   
Hit:6 https://box64.debian.ryanfortner.dev/debian ./ InRelease
Get:8 http://deb.debian.org/debian bookworm/main Sources [9,678 kB]
Ign:9 https://download.webmin.com/download/repository sarge InRelease
Hit:10 https://download.webmin.com/download/repository sarge Release
Get:11 http://deb.debian.org/debian bookworm/main arm64 Packages [8,892 kB]
Get:13 http://deb.debian.org/debian bookworm/main armhf Packages [8,679 kB]
Fetched 27.5 MB in 10s (2,710 kB/s)                                            
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
19 packages can be upgraded. Run 'apt list --upgradable' to see them.

Reinstalling apt dependencies...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 3 reinstalled, 0 to remove and 19 not upgraded.
Need to get 1,445 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 python3-dev arm64 3.11.2-1 [26.2 kB]
Get:2 http://deb.debian.org/debian bookworm/main arm64 python3-pip all 23.0.1+dfsg-1 [1,325 kB]
Get:3 http://deb.debian.org/debian bookworm/main arm64 python3-virtualenv all 20.17.1+ds-1 [93.9 kB]
Fetched 1,445 kB in 0s (4,891 kB/s)           
(Reading database ... 276642 files and directories currently installed.)
Preparing to unpack .../python3-dev_3.11.2-1_arm64.deb ...
Unpacking python3-dev (3.11.2-1) over (3.11.2-1) ...
Preparing to unpack .../python3-pip_23.0.1+dfsg-1_all.deb ...
Unpacking python3-pip (23.0.1+dfsg-1) over (23.0.1+dfsg-1) ...
Preparing to unpack .../python3-virtualenv_20.17.1+ds-1_all.deb ...
Unpacking python3-virtualenv (20.17.1+ds-1) over (20.17.1+ds-1) ...
Setting up python3-virtualenv (20.17.1+ds-1) ...
Setting up python3-dev (3.11.2-1) ...
Setting up python3-pip (23.0.1+dfsg-1) ...
Processing triggers for man-db (2.11.2-1) ...
Scanning processes...                                                           
Scanning candidates...                                                          
Scanning processor microcode...                                                 
Scanning linux images...                                                        

Running kernel seems to be up-to-date.

Failed to check for processor microcode upgrades.

Restarting services...
 systemctl restart wsdd.service
Service restarts being deferred:
 systemctl restart NetworkManager.service
 systemctl restart bluetooth.service
 /etc/needrestart/restart.d/dbus.service
 systemctl restart docker.service
 systemctl restart lightdm.service
 systemctl restart systemd-logind.service
 systemctl restart wpa_supplicant.service

No containers need to be restarted.
[... 132 lines cut ...]
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

Erasing any pre-existing virtual environment
Logging-out. You should login and re-run the menu.
Connection to europa closed.
ian@JupiterMint:~/IOTstack$

NossieUK, 2023-03-12:
**Soooooooooooooooooo

I installed bullseye on pi 4 4GB and upgraded to bookworm -  I then installed IOTstack from github and went through the whole process.  Effectively I'm still having the same issues.  I believe it's  debian or python imposing much more restrictive limitations on how pip performs  -- but unless your scripts are doping something extra dangerous  -- I really don't think there is much that can be done -  So   I'm posting this more as an FYI rather than something that can be easily fixed.**

pi@raspberrypi:~ $ curl -fsSL https://raw.githubusercontent.com/SensorsIot/IOTstack/master/install.sh | bash
IOTstack Installation
Enter in the sudo password when prompted, to install dependencies
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.39.2-1.1).
The following packages were automatically installed and are no longer required:
  agnostics edid-decode fio fonts-piboto freeglut3 g++-10 giblib1
  gnome-desktop3-data gnome-session-bin gnome-session-common
  gnome-settings-daemon-common gtk2-engines-clearlookspix gtk2-engines-pixflat
  gui-pkinst ibverbs-providers iio-sensor-proxy libaio1 libatk1.0-data
  libavresample4 libbasicusageenvironment1 libboost-iostreams1.74.0
  libboost-thread1.74.0 libcbor0 libcupsimage2 libdaxctl1 libegl-dev
  libevent-2.1-7 libflac8 libfluidsynth2 libfuse2 libgeoclue-2-0
  libgeocode-glib0 libgfapi0 libgfrpc0 libgfxdr0 libgl-dev libgl1-mesa-dev
  libgles-dev libglu1-mesa libglu1-mesa-dev libglusterfs0 libglut-dev
  libglut3.12 libglvnd-core-dev libglvnd-dev libglx-dev libgnome-desktop-3-20
  libgroupsock8 libgs9-common libgssdp-1.2-0 libgtksourceview-3.0-1
  libgtksourceview-3.0-common libgupnp-1.2-0 libgweather-3-16
  libgweather-common libibverbs1 libice-dev libicu67 libidn11 libilmbase25
  libjavascriptcoregtk-4.0-18 libjim0.79 libldap-2.4-2 liblivemedia77
  libllvm11 libmikmod3 libmms0 libmpdec3 libnbd0 libndctl6 libnma-common
  libnma0 libntfs-3g883 libofa0 libopenexr25 libopengl-dev libopengl0
  libpcre2-posix2 libperl5.32 libplacebo72 libpmem1 libpmemblk1 libpoppler102
  libprotobuf-lite23 libpthread-stubs0-dev libpulse-mainloop-glib0
  libpython3.9 libpython3.9-dev libpython3.9-minimal libpython3.9-stdlib
  libqpdf28 libqt5designer5 libqt5help5 libqt5test5 librados2 librbd1
  librdmacm1 librest-0.7-0 librtimulib-dev librtimulib-utils librtimulib7
  libsdl-mixer1.2 libsdl-ttf2.0-0 libsm-dev libsoup-gnome2.4-1
  libstdc++-10-dev libturbojpeg0 libusageenvironment3 libwebkit2gtk-4.0-37
  libx11-dev libxau-dev libxcb-res0 libxcb1-dev libxdmcp-dev libxext-dev
  libxfce4util-common libxfce4util7 libxfconf-0-3 libxkbregistry0 libxt-dev
  lxplug-bluetooth lxplug-cputemp lxplug-ejecter lxplug-magnifier lxplug-menu
  lxplug-netman lxplug-network lxplug-ptbatt lxplug-updater lxplug-volumepulse
  mutter-common perl-modules-5.32 pi-greeter pi-printer-support pipanel
  pishutdown pixflat-icons printer-driver-escpr python3-ldb python3-opengl
  python3-piexif python3-prctl python3-pyqt5 python3-pyqt5.sip python3-talloc
  python3-typed-ast python3-v4l2 python3.9 python3.9-dev python3.9-minimal
  qt5ct rp-bookshelf rpd-plym-splash rpd-wallpaper x11proto-dev xfconf
  xorg-sgml-doctools xtrans-dev xwayland
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
Cloning into 'IOTstack'...
remote: Enumerating objects: 9687, done.
remote: Counting objects: 100% (684/684), done.
remote: Compressing objects: 100% (264/264), done.
remote: Total 9687 (delta 371), reused 625 (delta 333), pack-reused 9003
Receiving objects: 100% (9687/9687), 17.06 MiB | 663.00 KiB/s, done.
Resolving deltas: 100% (5985/5985), done.
IOTstack cloned
Python Version: 'Python 3.11.2'. Python is outdated or virtualenv is missing
Installing Python3 and virtualenv
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-dev is already the newest version (3.11.2-1).
python3-dev set to manually installed.
The following packages were automatically installed and are no longer required:
  agnostics edid-decode fio fonts-piboto freeglut3 g++-10 giblib1
  gnome-desktop3-data gnome-session-bin gnome-session-common
  gnome-settings-daemon-common gtk2-engines-clearlookspix gtk2-engines-pixflat
  gui-pkinst ibverbs-providers iio-sensor-proxy libaio1 libatk1.0-data
  libavresample4 libbasicusageenvironment1 libboost-iostreams1.74.0
  libboost-thread1.74.0 libcbor0 libcupsimage2 libdaxctl1 libegl-dev
  libevent-2.1-7 libflac8 libfluidsynth2 libfuse2 libgeoclue-2-0
[... 366 lines cut ...]
      creating build/lib.linux-aarch64-cpython-311/ruamel/yaml
      copying .ruamel/yaml/__init__.py -> build/lib.linux-aarch64-cpython-311/ruamel/yaml
      creating build/lib.linux-aarch64-cpython-311/ruamel/yaml/clib
      copying ./__init__.py -> build/lib.linux-aarch64-cpython-311/ruamel/yaml/clib
      copying ./LICENSE -> build/lib.linux-aarch64-cpython-311/ruamel/yaml/clib
      running build_ext
      building '_ruamel_yaml' extension
      creating build/temp.linux-aarch64-cpython-311
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/IOTstack/.virtualenv-menu/include -I/usr/include/python3.11 -c _ruamel_yaml.c -o build/temp.linux-aarch64-cpython-311/_ruamel_yaml.o
[... cut ...]
            |            ^~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> ruamel.yaml.clib

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
pi@raspberrypi:~/IOTstack $

NossieUK, 2023-03-12:
And just as an additional - here is an attempt to install commanderpi with similar results

Anyway I'm leaving this here is a warning or a headsup to others - I really don't think there is a simple solution

`pi@raspberrypi:~/IOTstack $ cd
pi@raspberrypi:~ $ cd ~; git clone https://github.com/Jack477/CommanderPi
Cloning into 'CommanderPi'...
remote: Enumerating objects: 927, done.
remote: Counting objects: 100% (151/151), done.
remote: Compressing objects: 100% (101/101), done.
remote: Total 927 (delta 88), reused 105 (delta 50), pack-reused 776
Receiving objects: 100% (927/927), 342.14 KiB | 2.67 MiB/s, done.
Resolving deltas: 100% (567/567), done.
pi@raspberrypi:~ $ cd CommanderPi
pi@raspberrypi:~/CommanderPi $  sudo chmod +x install.sh
pi@raspberrypi:~/CommanderPi $ ./install.sh
/home/pi/CommanderPi
Checking for tkinter: Status: install ok installed
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.39.2-1.1).
python3-tk is already the newest version (3.11.2-2).
python3-pil is already the newest version (9.4.0-1.1+b1).
python3-pil set to manually installed.
The following packages were automatically installed and are no longer required:
  agnostics edid-decode fio fonts-piboto freeglut3 g++-10 giblib1 gnome-desktop3-data gnome-session-bin gnome-session-common gnome-settings-daemon-common gtk2-engines-clearlookspix gtk2-engines-pixflat gui-pkinst
  ibverbs-providers iio-sensor-proxy libaio1 libatk1.0-data libavresample4 libbasicusageenvironment1 libboost-iostreams1.74.0 libboost-thread1.74.0 libcbor0 libcupsimage2 libdaxctl1 libegl-dev libevent-2.1-7 libflac8
  libfluidsynth2 libfuse2 libgeoclue-2-0 libgeocode-glib0 libgfapi0 libgfrpc0 libgfxdr0 libgl-dev libgl1-mesa-dev libgles-dev libglu1-mesa libglu1-mesa-dev libglusterfs0 libglut-dev libglut3.12 libglvnd-core-dev
  libglvnd-dev libglx-dev libgnome-desktop-3-20 libgroupsock8 libgs9-common libgssdp-1.2-0 libgtksourceview-3.0-1 libgtksourceview-3.0-common libgupnp-1.2-0 libgweather-3-16 libgweather-common libibverbs1 libice-dev
  libicu67 libidn11 libilmbase25 libjavascriptcoregtk-4.0-18 libjim0.79 libldap-2.4-2 liblivemedia77 libllvm11 libmikmod3 libmms0 libmpdec3 libnbd0 libndctl6 libnma-common libnma0 libntfs-3g883 libofa0 libopenexr25
  libopengl-dev libopengl0 libpcre2-posix2 libperl5.32 libplacebo72 libpmem1 libpmemblk1 libpoppler102 libprotobuf-lite23 libpthread-stubs0-dev libpulse-mainloop-glib0 libpython3.9 libpython3.9-dev
  libpython3.9-minimal libpython3.9-stdlib libqpdf28 libqt5designer5 libqt5help5 libqt5test5 librados2 librbd1 librdmacm1 librest-0.7-0 librtimulib-dev librtimulib-utils librtimulib7 libsdl-mixer1.2 libsdl-ttf2.0-0
  libslirp0 libsm-dev libsoup-gnome2.4-1 libstdc++-10-dev libturbojpeg0 libusageenvironment3 libwebkit2gtk-4.0-37 libx11-dev libxau-dev libxcb-res0 libxcb1-dev libxdmcp-dev libxext-dev libxfce4util-common
  libxfce4util7 libxfconf-0-3 libxkbregistry0 libxt-dev lxplug-bluetooth lxplug-cputemp lxplug-ejecter lxplug-magnifier lxplug-menu lxplug-netman lxplug-network lxplug-ptbatt lxplug-updater lxplug-volumepulse
  mutter-common perl-modules-5.32 pi-greeter pi-printer-support pipanel pishutdown pixflat-icons printer-driver-escpr python3-ldb python3-opengl python3-piexif python3-prctl python3-pyqt5 python3-pyqt5.sip
  python3-talloc python3-typed-ast python3-v4l2 python3.9 python3.9-dev python3.9-minimal qt5ct rp-bookshelf rpd-plym-splash rpd-wallpaper slirp4netns x11proto-dev xfconf xorg-sgml-doctools xtrans-dev xwayland
Use 'sudo apt autoremove' to remove them.
Suggested packages:
  python-pil-doc
The following NEW packages will be installed:
  python3-pil.imagetk
0 upgraded, 1 newly installed, 0 to remove and 5 not upgraded.
Need to get 74.4 kB of archives.
After this operation, 159 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://deb.debian.org/debian bookworm/main arm64 python3-pil.imagetk arm64 9.4.0-1.1+b1 [74.4 kB]
Fetched 74.4 kB in 0s (417 kB/s)             
Selecting previously unselected package python3-pil.imagetk:arm64.
(Reading database ... 112004 files and directories currently installed.)
[... 7 lines cut ...]
Running kernel seems to be up-to-date.

Failed to check for processor microcode upgrades.

No services need to be restarted.

No containers need to be restarted.

No user sessions are running outdated binaries.

No VM guests are running outdated hypervisor (qemu) binaries on this host.
/home/pi/CommanderPi
[Desktop Entry]
Name=CommanderPi
Comment=System info and overclocking
Exec=/home/pi/CommanderPi/src/start.sh
Icon=/home/pi/CommanderPi/src/icons/Icon.png
Categories=Utility;
Version=1.0
Type=Application
Terminal=false
StartupNotify=true
Save desktop shortcut to /home/pi/Desktop/commanderpi.desktop
Save menu shortcut to /usr/share/applications/commanderpi.desktop
`

Paraphraser, 2023-03-12:
So, what happens if you switch to the old-menu branch in IOTstack and try to run the "old menu"?

Maybe that's a workaround for someone who wants to run Bookworm + IOTstack. These days, the only real difference between the two is *some* service definitions still use environment files in the services directory. Networking (which used to be a big difference) is harmonised. There's no good reason to pick old or new.

NossieUK, 2023-03-12:
I changed my production stack install to the old menu and it's working fine.  Never thought of doing that before - but as I now only use iot stack for the handy menu and manually code the compose file myself - this is a great work around for me =)

I'm just surprised more people are not getting trapped in this, it's like since bookworm they have suddenly enforced dependency rules -- and I don't want to override it by default.

`  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for ruamel.yaml.clib
  error: subprocess-exited-with-error
  
  × Running setup.py install for ruamel.yaml.clib did not run successfully.
  │ exit code: 1
  ╰─> [27 lines of output]
      sys.argv ['/tmp/pip-install-inqdcts6/ruamel-yaml-clib_f4410fc5a3d74295837bbb5725fe74f2/setup.py', 'install', '--record', '/tmp/pip-record-fe1bcw30/install-record.txt', '--single-version-externally-managed', '--compile', '--install-headers', '/home/pi/IOTstack/.virtualenv-menu/include/site/python3.11/ruamel.yaml.clib']
      test compiling /tmp/tmp_ruamel_sy25bxg8/test_ruamel_yaml.c -> test_ruamel_yaml OK
      /home/pi/IOTstack/.virtualenv-menu/lib/python3.11/site-packages/setuptools/dist.py:285: SetuptoolsDeprecationWarning: The namespace_packages parameter is deprecated, consider using implicit namespaces instead (PEP 420).
        warnings.warn(msg, SetuptoolsDeprecationWarning)
      running install
      /home/pi/IOTstack/.virtualenv-menu/lib/python3.11/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-311
      creating build/lib.linux-aarch64-cpython-311/ruamel
      copying .ruamel/__init__.py -> build/lib.linux-aarch64-cpython-311/ruamel
      creating build/lib.linux-aarch64-cpython-311/ruamel/yaml
      copying .ruamel/yaml/__init__.py -> build/lib.linux-aarch64-cpython-311/ruamel/yaml
      creating build/lib.linux-aarch64-cpython-311/ruamel/yaml/clib
      copying ./__init__.py -> build/lib.linux-aarch64-cpython-311/ruamel/yaml/clib
      copying ./LICENSE -> build/lib.linux-aarch64-cpython-311/ruamel/yaml/clib
      running build_ext
      building '_ruamel_yaml' extension
      creating build/temp.linux-aarch64-cpython-311
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/IOTstack/.virtualenv-menu/include -I/usr/include/python3.11 -c _ruamel_yaml.c -o build/temp.linux-aarch64-cpython-311/_ruamel_yaml.o
      _ruamel_yaml.c:198:12: fatal error: longintrepr.h: No such file or directory
        198 |   #include "longintrepr.h"
            |            ^~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> ruamel.yaml.clib

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.`

ah well, I'll revisit again another time - but thanks for your thoughts and for giving me work around

NossieUK, 2023-03-21:
So I think I finally got to the bottom of these errors - but I've no idea when it was added as for a good 6 months it worked great

https://www.debian.org/releases/bookworm/arm64/release-notes/ch-information.en.html#python3-pep-668

5.2.2. Python Interpreters marked externally-managed

The Debian provided python3 interpreter packages (python3.11 and pypy3) are now marked as being externally-managed, following [PEP-668](https://peps.python.org/pep-0668/). The version of python3-pip provided in Debian follows this, and will refuse to manually install packages on Debian's python interpreters, unless the --break-system-packages option is specified.

If you need to install a Python application (or version) that isn't packaged in Debian, we recommend that you install it with pipx (in the pipx Debian package). pipx will set up an environment isolated from other applications and system Python modules, and install the application and its dependencies into that.

If you need to install a Python library module (or version) that isn't packaged in Debian, we recommend installing it into a virtualenv, where possible. You can create virtualenvs with the venv Python stdlib module (in the python3-venv Debian package) or the virtualenv Python 3rd-party tool (in the virtualenv Debian package). For example, instead of running pip install --user foo, run: mkdir -p ~/.venvs && python3 -m venv ~/.venvs/foo && ~/.venvs/foo/bin/python -m pip install foo to install it in a dedicated virtualenv.

See /usr/share/doc/python3.11/README.venv for more details.

NossieUK, 2023-07-02:
Just as an FYI - I accidently hosed my OS trying to update from bookworm to trixie  on the pi4

So copied everything I need off and I'm currently in the process of going back to bookworm.

As bookworm is about to get released officially for Pi in the next month or two - I just want to highlight this issue once more.   I'm pretty sure it's something to do with how the virtual environment is configured for IOTstack but I'm still getting the same error with a brand new install of bullseye and then a straight upgrade to bookworm - so this issue is getting closer and closer to  mainstream users.

Just a heads up though

`
You should now restart your system
Setting up environment:
Kernel cgroups not enabled. Adding kernel parameters.
You will need to restart your system before the changes take effect.
console=serial0,115200 console=tty1 root=PARTUUID=faca963f-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cgroup_memory=1 cgroup_enable=memory
pi@Europa:~ $ sudo reboot

Broadcast message from root@Europa on pts/1 (Sun 2023-07-02 14:09:56 BST):

The system will reboot now!

pi@Europa:~ $

[SSH] INFO: DISCONNECT

[SSH] Server Version OpenSSH_9.2p1 Debian-2
[SSH] Encryption used: chacha20-poly1305@openssh.com
[SSH] Logged in (password)

Linux Europa 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Jul  2 14:10:15 2023
pi@Europa:~ $ cd IOTstack/
pi@Europa:~/IOTstack $ ./menu.sh
Checking for project update
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
Project is up to date
Python virtualenv found.
Python Version: 'Python 3.11.2'. Python and virtualenv is up to date.
Please enter sudo pasword if prompted
Command: docker version -f "{{.Server.Version}}"
Docker version 20.10.24+dfsg1 >= 18.2.0. Docker is good to go.
Project dependencies up to date

Creating python virtualenv for menu...
Installing menu requirements into the virtualenv...
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [25 lines of output]
      sys.argv ['/tmp/pip-install-z5ll71x5/ruamel-yaml-clib_a9691471879645fe96d563ce096bef92/setup.py', 'bdist_wheel', '-d', '/tmp/pip-wheel-70n3i0ah']
      test compiling /tmp/tmp_ruamel_xkeuxpey/test_ruamel_yaml.c -> test_ruamel_yaml OK
      /home/pi/IOTstack/.virtualenv-menu/lib/python3.11/site-packages/setuptools/dist.py:285: SetuptoolsDeprecationWarning: The namespace_packages parameter is deprecated, consider using implicit namespaces instead (PEP 420).
        warnings.warn(msg, SetuptoolsDeprecationWarning)
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-311
      creating build/lib.linux-aarch64-cpython-311/ruamel
      copying .ruamel/__init__.py -> build/lib.linux-aarch64-cpython-311/ruamel
      creating build/lib.linux-aarch64-cpython-311/ruamel/yaml
      copying .ruamel/yaml/__init__.py -> build/lib.linux-aarch64-cpython-311/ruamel/yaml
[... 34 lines cut ...]
      creating build/lib.linux-aarch64-cpython-311/ruamel/yaml
      copying .ruamel/yaml/__init__.py -> build/lib.linux-aarch64-cpython-311/ruamel/yaml
      creating build/lib.linux-aarch64-cpython-311/ruamel/yaml/clib
      copying ./__init__.py -> build/lib.linux-aarch64-cpython-311/ruamel/yaml/clib
      copying ./LICENSE -> build/lib.linux-aarch64-cpython-311/ruamel/yaml/clib
      running build_ext
      building '_ruamel_yaml' extension
      creating build/temp.linux-aarch64-cpython-311
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/IOTstack/.virtualenv-menu/include -I/usr/include/python3.11 -c _ruamel_yaml.c -o build/temp.linux-aarch64-cpython-311/_ruamel_yaml.o
      _ruamel_yaml.c:198:12: fatal error: longintrepr.h: No such file or directory
        198 |   #include "longintrepr.h"
            |            ^~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> ruamel.yaml.clib

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
pi@Europa:~/IOTstack $                                          `

Paraphraser, 2023-07-03:
I think I have found the cause. See #723 for the explanation.

I have also added some modifications to PiBuilder to support Bookworm. See the PiBuilder [Change Log](https://github.com/Paraphraser/PiBuilder/blob/master/CHANGELOG.md) for the details.

Here's a Pi 4 happily running IOTstack on Bookworm:

https://github.com/SensorsIot/IOTstack/assets/34226495/846d4dc6-8ab4-4660-8a46-21eb9f4302e2

<hr>

By the way, this line in your output troubles me:

```
Docker version 20.10.24+dfsg1 >= 18.2.0. Docker is good to go.
```

I think of that "+dfsg1" as the signature of an incorrect installation of `docker` and `docker-compose`. I believe the sequence that leads to it is:

1. Install `docker` using the convenience script.
2. Install `docker-compose` via `sudo apt install docker-compose`.

For some reason, using `apt` to install `docker-compose` gets version 1.29.2 (ancient) and that, in turn, somehow results in `docker` downgrading from whatever the convenience script installed (currently 24.0.2) to 20.10.24 with that "+dfsg1" suffix.

I'm aware that both `install.sh` and `scripts/install_docker.sh` install `docker-compose` using `apt` (and create this problem) but they are wrong.

The correct way to install `docker` is just step 1. That gets `docker` and `docker-compose-plugin` which gets you the `docker compose` command (no hyphen). Then, if you want the hyphenated form (which you probably do else every script that invokes `docker-compose` will fail), you symlink `/usr/local/bin/docker-compose` to `/usr/libexec/docker/cli-plugins/docker-compose`. That's what the PiBuilder [`04_setup.sh`](https://github.com/Paraphraser/PiBuilder/blob/99820adde19bb22faa72d58ce98b6fcc7b035256/boot/scripts/04_setup.sh#L73) does.

The version of `docker-compose-plugin` installed by the convenience script is 2.18.1. Once both `docker` and `docker-compose-plugin` have been installed correctly, they are kept up-to-date with routine `apt upgrade`. Conversely, once they have been installed incorrectly, they are stuck that way for all time.

On a number of occasions I've thought about preparing a pull request to try to fix the IOTstack scripts. Each time I've thrown up my hands in horror and decided that I really don't need either the headache of trying to sort it out, or the (implied) responsibility for future maintenance. I'm already doing PiBuilder and I reckon that's enough.

PR #723 (2023-09-23): 2023-07-03 Python requirements files + Bookworm - master branch
The underlying cause of the problem reported in #660 is the Python requirements files containing pins for (almost) every line item.

I appreciate that pinning is what happens by default when requirements files are generated but some of the pinned versions which might have been appropriate for Bullseye when the files were generated will not actually install on Bookworm.

This Pull Request removes pins from every line item in each requirements file. In other words, the defaults will become the latest and greatest for each system (Bu [... cut]
commit 611f70ec (2023-09-23): 2023-07-03 Python requirements files + Bookworm - master branch

The underlying cause of the problem reported in #660 is the Python
requirements files containing pins for (almost) every line item.

I appreciate that pinning is what happens by default when requirements
files are generated but some of the pinned versions which might have
been appropriate for Bullseye when the files were generated will not
actually install on Bookworm.

This Pull Request removes pins from every line item in each requirements
file. In other words, the defaults will become the latest and greatest
for each system (B [... cut]
closed 2023-09-23 by Slyke