SensorsIot/IOTstack #677: Python 3.6.9 or later (Current = 3.7.3) and virtualenv (Installed = false) are required

dokitoki, 2023-04-05:
It looks like the virtualenv commit for menu.sh has some problems with newer version of python.

Env: Raspberry Pi with 
Linux raspberrypi3 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
Python 3.7.3 

After running ./menu.sh the error is 
```
| Python 3.6.9 or later (Current = 3.7.3) and virtualenv (Installed = false) │ 
│ are required for IOTstack to function correctly. Install these now?          |

```
After selecting to install:

```
Python Version: 'Python 3.7.3'. Python is outdated or virtualenv is missing
....
python3-dev is already the newest version (3.7.3-1).
python3-virtualenv is already the newest version (15.1.0+ds-2+deb10u1).
....
Existing installation detected.
Creating python virtualenv for menu...
./menu.sh: line 396: virtualenv: command not found

```
Also:

```
> pip3 install virtualenv
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: virtualenv in /usr/lib/python3/dist-packages (15.1.0)

```

Is this because after Python 3.7 virutalenv is used via "python3 -m virtualenv"?

Noschvie, 2023-04-05:
Same issue …

Paraphraser, 2023-04-05:
Am I correct that this is a Raspberry Pi 3?

I don't really understand how to interpret all the fine nuances of `uname -a` output but when I do it on a Pi4 running full 64-bit Bullseye (that's all I have to test on):

```
$ uname -a
Linux iot-hub 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux
```

I look at the `5.10.103-v7l+` in yours and the `6.1.19-v8+` and wonder if that means Buster?

The reason I ask is because the script I'm about to point you to has been tested on Raspberry Pi OS Bullseye but not on Buster. It might work but it might make things worse.

It sounds like this might be the same as #601. If you agree, click [this link](https://github.com/Paraphraser/PiBuilder/blob/master/reset_menu_enviroment.md) and follow the instructions.

Please let me know the outcome.

Noschvie, 2023-04-05:
> It sounds like this might be the same as #601. If you agree, click [this link](https://github.com/Paraphraser/PiBuilder/blob/master/reset_menu_enviroment.md) and follow the instructions.

This worked for me too, thanks @Paraphraser

400bird, 2023-06-25:
I was getting the same exact messages, the Pi builder script linked above did not change the situation.
Following the link to #601, this worked for me:

```
$ sudo apt install -y python3-pip python3-dev python3-virtualenv
$ sudo pip3 install -U virtualenv
```

johncaipa, 2024-03-16:
I have the same problem but the commands mentioned above have not worked, I have the following rpi system:
Linux edgebox 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux

error:
[image]
[image]

Paraphraser, 2024-03-17:
@johncaipa Someone with more Python knowledge than I have may want to correct me but, when I look at your `uname -a` output, I see a very old 32-bit system (2021, armv7l). Is that Buster?

The mention of 307 packages needing updating also gives me the frights. Running an `apt upgrade` will change an awful lot of things at the same time, which is rarely an entirely safe thing to be doing with any computer system.

When I see "old", I always start to wonder about the rigorousness with which developers maintain backwards compatibility. I'm not saying that's what's going on here with Python, merely that that *might* explain it.

Before you decide to rebuild using 64-bit Bookworm, perhaps try this:

```
$ git clone https://github.com/SensorsIot/IOTstack.git ~/IOTstackClean
$ cd ~/IOTstackClean
$ ./menu.sh
```

If that doesn't work then (a) read to the end of this reply before you do anything, (b) take a backup of your IOTstack and (c) cross your fingers, hold your breath and run:

```
$ sudo apt update
$ sudo apt upgrade -y
$ sudo apt autoremove -y
```

and then try the menu again, perhaps also with the virtual-env stuff above.

If you can't get anywhere (or, worst case, the upgrade breaks your system) then I think you'll need to rebuild. If you decide to do that, I recommend [PiBuilder](https://github.com/Paraphraser/PiBuilder). I've just tested it with the 2024-03-12 Bookworm releases, including using [IOTstackBackup](https://github.com/Paraphraser/IOTstackBackup) to restore backups taken on a Bullseye system.

The PiBuilder README may look a bit daunting but it boils down to:

1. Download the up-to-date Raspbian image.
2. Use Raspberry Pi Imager to burn to SD/SSD.
3. Boot the Pi and connect over SSH.
4. Clone PiBuilder onto the Pi.
5. Run PiBuilder's five scripts, in order.
6. Restore the backup.
7. Bring up the stack.

Please ask if you need any help with any of this. In particular, if you aren't using IOTstackBackup already, you need to be slightly cautious about how you prepare the backup. In essence, your stack needs to be down, you need to save `docker-compose.yml`, `.env` (if it exists), any override files you might be using, the `services` directory and the `volumes` directory. With `volumes` it's important to preserve ownership and permissions so they can be restored properly later.

Also, the combination of 2021, armv7l and 307 packages makes me wonder whether you have kept your Docker images up-to-date? Not every Docker image likes jumping multiple versions so, after a rebuild, a "restore" and an "up", you might find containers moaning.

Sorry to be the bearer of bad news like this but…