SensorsIot/IOTstack #688: Nodered setup missing file.

theDiverDK, 2023-04-19:
I did a clean install yesterday.

And when i select 'Nodered' i get this error in the bottom of the menu:

nodered (fileIssues) - ['/addons_list.yml does not exist. Build addons file in options to fix. This is optional']

and when pressing 'enter' to setup the docker-compose file i get this python error:

FileNotFoundError: [Errno 2] No such file or directory: './services/nodered/addons_list.yml'

The folder exists, but the file does not

If i try to start the containers with:
docker-compose up -d

I get this error:
WARNING: Some networks were defined but are not used by any service: nextcloud
Building nodered
ERROR: Cannot locate specified Dockerfile: Dockerfile

Have i forgotten to do something?
I follow the automatic guide here:
https://sensorsiot.github.io/IOTstack/Basic_setup/

Paraphraser, 2023-04-19:
When you run the menu and select Node-RED, you need to press the right-arrow to select the add-on nodes you want.

There is a prompt about it but it's not *completely* obvious what it means (the prompt is *trying* to say "pay attention to the double-right-pointing arrows with the word Options" where the double-right-pointing arrows mean "press right arrow"). And it also suggests the process is *optional* when, in fact, it's *mandatory*:

[image]

So, re-run the menu, put the cursor on top of nodered, press right-arrow, the cursor will land on "Select & build addons list", and press return. That will take you to a list of add-on nodes, some of which are selected already. If you know you are going to need some specific add-on nodes that aren't included by default then find them and select them. Note that this list is not exhaustive so there may be nodes you need that aren't mentioned in this list. Once you've finished making selections, press return (which generates the Dockerfile - the thing that was reported as missing above). Then select "go back" which takes you back to the "build stack" menu. Hit enter again (which generates the docker-compose.yml).

After that, a `docker-compose up -d` will pull down the base image for Node-RED from DockerHub, the Dockerfile will run to do a local build to include add-ons, and the nodered container will come up.

Just as a heads-up, although you **can** re-run the menu to change the list of add-on nodes, it's usually both simpler and safer to edit the Dockerfile with a text editor. Here's the version you get if you just accept the default add-on nodes:

```
$ cat ~/IOTstack/services/nodered/Dockerfile 
# reference argument - omitted defaults to latest
ARG DOCKERHUB_TAG=latest

# Download base image
FROM nodered/node-red:${DOCKERHUB_TAG}

# reference argument - omitted defaults to null
ARG EXTRA_PACKAGES
ENV EXTRA_PACKAGES=${EXTRA_PACKAGES}

# default user is node-red - need to be root to install packages
USER root

# install packages
RUN apk update && apk add --no-cache eudev-dev ${EXTRA_PACKAGES}

# switch back to default user
USER node-red

# variable not needed inside running container
ENV EXTRA_PACKAGES=

# add-on nodes follow

RUN cd /usr/src/node-red && npm install --save  node-red-configurable-ping
RUN cd /usr/src/node-red && npm install --save  node-red-contrib-boolean-logic
RUN cd /usr/src/node-red && npm install --save  node-red-contrib-influxdb
RUN cd /usr/src/node-red && npm install --save  node-red-dashboard
RUN cd /usr/src/node-red && npm install --save  node-red-node-pi-gpiod
RUN cd /usr/src/node-red && npm install --save  node-red-node-rbe
```

If you want to add a package, just follow the pattern of those lines at the end, and then run:

```
$ cd ~/IOTstack
$ docker-compose up --build -d nodered
$ docker system prune -f
```

The `--build` causes the Dockerfile to be re-run. The `prune` cleans up the old image.

There's a bit more information in the [Node-RED wiki](https://sensorsiot.github.io/IOTstack/Containers/Node-RED/#componentManagement). In practice, editing the Dockerfile is the most robust method. Although you can use Manage Palette, it usually results in duplicates where older versions are blocking newer versions, and then you have a mess to sort out (not difficult, just an annoyance).

Some of this may sound like it's all a bit half-baked but it's sort of the result of "dockerising" processes like Node-RED. There are compromises. The key benefit, however, of being able to upgrade a container, roll back, and otherwise treat it as a self-contained unit is well worth the price of a bit of mental strain until you get the hang of it.

theDiverDK, 2023-04-19:
OMG what an AMAZING answer.

Thanks a lot.

Glad to see, it was just me and not some bug.

I'll follow your advice soon :)

Thanks

theDiverDK, 2023-04-19:
It works, THANKS :)

closed 2023-04-19 by theDiverDK