SensorsIot/IOTstack #720: Problem with password to webGui

tomuszom, 2023-06-27:
After each uprade (release new version) of pihole (docker-compose pull / docker-compose up -d) my password not working. 
env is set in file

pi@raspberrypi:~/IOTstack $ cat services/pihole/pihole.env
TZ=Europe/Warsaw
WEBPASSWORD=HbH9lFE4

Paraphraser, 2023-06-27:
Does your Pi-hole service definition in your `docker-compose.yml` look like this (from old-menu branch)?

``` yaml  
  pihole:
    container_name: pihole
    image: pihole/pihole:latest
    ports:
      - "53:53/tcp"
      - "53:53/udp"
      - "67:67/udp"
      - "8089:80/tcp"
    env_file:
      - ./services/pihole/pihole.env
    volumes:
      - ./volumes/pihole/etc-pihole:/etc/pihole
      - ./volumes/pihole/etc-dnsmasq.d:/etc/dnsmasq.d
    dns:
      - 127.0.0.1
      - 1.1.1.1
    cap_add:
      - NET_ADMIN
    restart: unless-stopped
```

or this (from master branch):

``` yaml
  pihole:
    container_name: pihole
    image: pihole/pihole:latest
    ports:
      - "8089:80/tcp"
      - "53:53/tcp"
      - "53:53/udp"
      - "67:67/udp"
    environment:
      - TZ=${TZ:-Etc/UTC}
      - WEBPASSWORD=
      # see https://sensorsiot.github.io/IOTstack/Containers/Pi-hole/#adminPassword
      - INTERFACE=eth0
      - FTLCONF_MAXDBDAYS=365
      - PIHOLE_DNS_=8.8.8.8;8.8.4.4
      # see https://github.com/pi-hole/docker-pi-hole#environment-variables
    volumes:
      - ./volumes/pihole/etc-pihole:/etc/pihole
      - ./volumes/pihole/etc-dnsmasq.d:/etc/dnsmasq.d
    dns:
      - 127.0.0.1
      - 1.1.1.1
    cap_add:
      - NET_ADMIN
    restart: unless-stopped
```

It might not look like it's either but I'm really interested in these lines:

* from old-menu branch

	```
	    env_file:
	      - ./services/pihole/pihole.env
	```

* from master branch

	```
	    environment:
	      - TZ=${TZ:-Etc/UTC}
	      - WEBPASSWORD=
	```

If your service definition contains the `env_file:` clause then what you are doing should be working.

However, if your service definition contains the `environment:` clause then docker-compose is not reading your `services/pihole/pihole.env`. I'm going to assume this is the most likely problem.

What I recommend you do is start by making your service definition look like the one from "master branch".

Next, do this:

```
$ cd ~/IOTstack
$ cat .env
```

If you get an error, or if you get output that doesn't contain:

```
Europe/Warsaw
```

then fix it with:

```
$ echo "TZ=Europe/Warsaw" >>.env
```

> We have been working through the service definitions to try to make sure they all use `- TZ=${TZ:-Etc/UTC}`. That syntax means you only have to set the TZ once in the `~/IOTstack/.env` file so this is as good a time as any to set it up properly.

If you have any other environment variables in your `services/pihole/pihole.env` that you haven't shown in your question then bring those over too.

Now, as for the password, the current service definitions have `WEBPASSWORD=` so it's a null string. That means PiHole won't have any password at all (that's how I run it myself).

If you want a password then add it:

```
- WEBPASSWORD=HbH9lFE4
```

Each time you "up" the container, the value of `WEBPASSWORD` is enforced. It's exactly like the old `-a -p` command.

Hope this helps. If you keep having problems then post your service definition here and I'll take another look.

tomuszom, 2023-07-02:
Thanks, found issue. There was WEBPASSWORD in docker-compose.yml that was different that I always used. Changed and it's fine right now.

closed 2023-07-02 by tomuszom