SensorsIot/IOTstack #727: zigbee2mqtt not population volumes/zigbee2mqtt/data

MarvinWeitz, 2023-09-23:
Hi there,

I wanted to install my Home Assistant setup from scratch using mosquitto and zigbee2mqtt.
After choosing both in my menu, zigbee2mqtt ends up in a restart loop without wrting logs to volumes/zigbee2mqtt/data. I accessed the logs via portainer and ended up with the following:
```
Using '/app/data' as data directory
Creating configuration file...
cp: can't stat '/app/configuration.example.yaml': No such file or directory
```
Upon inspection, the data folder indeed didn't have a config file present. Is there an update I missed here?

Paraphraser, 2023-09-24:
See [Koenkk/zigbee2mqtt/issues/19062](https://github.com/Koenkk/zigbee2mqtt/issues/19062) which I submitted a few moments ago.

This is not an IOTstack problem, as such. It seems to be a very recent problem with the Zigbee2MQTT image on DockerHub.

If you read the issue linked above, you'll see that there's a workaround. Just start the container with an explicit tag of `1.32.2`. That will initialise the persistent store properly. Then change the tag back to `latest`.

I'm embarrassed to admit that I ran into this exact same problem a day or two ago. I was testing the process of getting a Zigbee adapter connected to an old Intel-based Mac running Proxmox with a Debian guest. I saw this and assumed it was something I had done wrong. I knew Zigbee2MQTT was running on a Pi at `latest` so I just cloned the `~/IOTstack/volumes/zigbee2mqtt` folder from the Pi onto the Proxmox host. The problem went away so I went back to the task at hand (documenting the how-to) and didn't think much more about it. My bad.

Anyway, thanks for reporting this. You have the workaround so anyone else hitting the same problem will (hopefully) find your issue and it will lead them in the right direction.

> That's actually a strong hint for you to please **NOT** close your issue until it is solved in a later release of Zigbee2MQTT. The trick is to place a watch on issue 19062 so you'll get a notification when it is closed, then you can close this one. I apologise if I'm telling you things you already know.

MarvinWeitz, 2023-09-24:
@Paraphraser thanks so much for the answer. Using the older version fixed it. I'll keep this open as you suggested.

closed 2023-10-02 by MarvinWeitz