SensorsIot/IOTstack #730: How-to: Two Node-Red Containers in parallel

Noschvie, 2023-10-04:
Hello
To service two Node-Red dashboards independently I need to run two Node-Red containers in parallel hosted by a Raspberry.

How do that ?
(... with different port numbers)
Running one IOTStack and duplicating the Node-Red service to noderedA and noderedB ?
Running two IOTStacks in parallel using directory IOTstackA and IOTstackB ?

Thanks!

Slyke, 2023-10-04:
Get a new instance up and running, edit the `docker-compose.yaml` file, copy the `nodered` block, paste it, and ensure to change its name from `nodered` to something else.

Remember to change the volumes that it mounts, and also the port.

Restart IOTStack.

Example:
```
nodered-2:
  container_name: nodered
  build:
    context: ./services/nodered/.
    args:
      - DOCKERHUB_TAG=latest
      - EXTRA_PACKAGES=
  restart: unless-stopped
  user: "0"
  environment:
    - TZ=${TZ:-Etc/UTC}
  ports:
    - "1880:1880"
  volumes:
    - ./volumes/nodered2/data:/data
    - ./volumes/nodered2/ssh:/root/.ssh
 ```

Ensure the `nodered2` (or whatever you call it) directories exist. You may just want to copy the existing nodered directories and rename it from there.

Let us know if you run into errors!

Paraphraser, 2023-10-04:
And my 2¢ is slightly different from Slyke's

<hr>

The default service definition is:

``` yaml
nodered:
  container_name: nodered
  build:
    context: ./services/nodered/.
    args:
      - DOCKERHUB_TAG=latest
      - EXTRA_PACKAGES=
  restart: unless-stopped
  user: "0"
  environment:
    - TZ=${TZ:-Etc/UTC}
  ports:
    - "1880:1880"
  volumes:
    - ./volumes/nodered/data:/data
    - ./volumes/nodered/ssh:/root/.ssh
```

When you run the menu, choose nodered and hit the right-arrow to select your add-on nodes, some files are copied from the template into `~/services/nodered`. The important one is the Dockerfile which includes the nodes you have selected.

To run a second instance of Node-RED in parallel:

1. Make a copy of the "services" directory:

	``` console
	$ cd ~/IOTstack/services
	$ cp -a nodered nodered2
	```
	
2. If you need **different** add-on nodes in the second instance then use a text editor to edit:

	```
	~/IOTstack/services/nodered2/Dockerfile
	```

3. Use a text editor to edit:

	```
	~/IOTstack/docker-compose.yml
	```

	The edits are:
	
	1. Clone the existing nodered service definition.
	2. Change all mentions of "nodered" to "nodered2".
	3. Select a different external port (I'm going to use 1881).

	The result will be something like this:
	
	``` yaml
	nodered2:
	  container_name: nodered2
	  build:
	    context: ./services/nodered2/.
	    args:
	      - DOCKERHUB_TAG=latest
	      - EXTRA_PACKAGES=
	  restart: unless-stopped
	  user: "0"
	  environment:
	    - TZ=${TZ:-Etc/UTC}
	  ports:
	    - "1881:1880"
	  volumes:
	    - ./volumes/nodered2/data:/data
	    - ./volumes/nodered2/ssh:/root/.ssh
	```
	
4. If you do **nothing** else then the second instance of Node-RED will build and then initialise itself as a blank slate (no flows).

	However, if you want the second instance to start off as a clone of your existing version then, **before** you start the new instance for the first time, also do:
	
	``` console
	$ cd ~/IOTstack/volumes
	$ sudo cp -a nodered nodered2
	```
	
	**BUT** please be aware that if any of your existing flows writes to a database (eg InfluxDB) then both of your Node-RED instances may well wind up writing the same records to the database so you'll wind up with duplications.
	
	In other words, please don't just say "heck-yes I'll start with a perfect clone and then go tinkering". Please consider whether a perfect clone is likely to result in unintended consequences.
	
	> I give this caution having made this very mistake. It was a pain to unpick.
	
5. When you are ready:

	```
	$ cd ~/IOTstack
	$ docker-compose up -d
	```

Final caution. The IOTstack menu won't have any idea about what you've done so don't run the menu again (at least not without making a backup copy of your compose file).

Noschvie, 2023-10-04:
Have duplicated the nodered service definition to nodered2 and also the volume directory to nodered2.
Both Node-Red instances are up-and-running.
Now I have to check the backup script...
Thanks again!

closed 2023-10-04 by Noschvie

Paraphraser, 2023-10-04:
Well, if you're using IOTstackBackup, it will capture the second instance of Node-RED as part of the general backup.

I don't use the backup provided with IOTstack but I'm pretty sure it will too because it just expects you to down the stack and then grabs everything.