SensorsIot/IOTstack #736: Grafana installation issue

SHaque0, 2023-10-13:
Hello, 

I am trying to install Grafana, but when i'm trying to start the stack it is throwing an error: 
```ERROR: no matching manifest for linux/arm/v7 in the manifest list entries```

Hardware: Raspberry Pi 4B 4gb
OS: Raspberry Pi OS Lite 64-bit

Thanks in advance.

Paraphraser, 2023-10-13:
Try again. The [DockerHub](https://hub.docker.com/r/grafana/grafana/tags) page looks like it has been updated more recently than your post. I just did a "pull" on a Pi 4B (64-bit Bullseye) and version 10.1.5 of Grafana came down successfully.

This is not the first time this has happened (and not just with Grafana). I am not sure of the exact cause but I think it must be possible for a build to only succeed partially (eg the build of Grafana for AMD64 succeeds while the build for ARM64 fails), and then what gets pushed to DockerHub only includes the one architecture. Then "the world screams", someone takes a look, fixes the problem, triggers a rebuild, and the problem goes away. Just speculation but it fits the pattern.

You can usually get a handle on this by going to the DockerHub page for the image that's giving you a hassle and eyeball what's just gone up vs what was there for prior versions.

SHaque0, 2023-10-25:
it worked

gadhvirushiraj, 2023-12-19:
Hello,
I am having the same issue with Raspberry Pi 3B+  boards, any ideas ?

Paraphraser, 2023-12-19:
This [question](https://discord.com/channels/638610460567928832/638610461109256194/1185690960991105194) was also asked on the IOTstack Discord channel. I'm re-posting my [answer](https://discord.com/channels/638610460567928832/638610461109256194/1185712802023874701) here.

But please make sure you also read the addendum.

<hr>

[image]

The screen shot is from [DockerHub](https://hub.docker.com/r/grafana/grafana/tags) (specifically, the second page of that URL). Notice how the `10.1.2` tag has a `linux/arm/v7` architecture while `10.1.4` doesn't. If you nose around you'll see that the pattern is as I've shown with the green and red arrows.

The basic problem is that your Raspberry Pi is not running a full 64-bit operating system. It's either a full 32-bit version of Raspberry Pi OS or a mixed-mode where the kernel is 64-bit while the user space is 32-bit.

The only way your Raspberry Pi (the hardware) can run Grafana 10.1.4 or later is if you upgrade to a full 64-bit OS. The most reliable way to do that is to start with a clean image from [Raspberry Pi OS](https://www.raspberrypi.com/software/operating-systems/) where you would choose one of the options under "Raspberry Pi OS (64-bit)".

The alternative is to keep running your current system but pin Grafana to 10.1.2. You do that by editing Grafana's service definition in your `docker-compose.yml` to look like this:

``` yaml
  grafana:
    container_name: grafana
    image: grafana/grafana:10.1.2
    restart: unless-stopped
    …
```

Then do a `docker-compose up -d`. Pinning the version at least gets Grafana running again, and gives you time to decide when/if you want to rebuild on top of a 64-bit base. As you contemplate that, please keep in mind that this problem (of images no longer supporting 32-bit mode) is only going to get worse. Plus, Grafana is an example of one of the "good guys" where DockerHub has historical images which you can pin to. Some repos don't do that and then you have to resort to inventing your own local tags so you can pin to those.

In short: (a) start planning and (b) don't let your plan gather too much dust. 🤓 

If you're wondering, there's basically no good reason not to run full 64-bit Bookworm. Even if you are using containers which **only** have `linux/arm/v7` architectures, those will still work in a 64-bit OS.

<hr>

## Addendum

Just to be clear, the actual problem that led to the original question when this issue was first opened turned out to be a glitch in the release process (ie the omission of the `linux/arm/v7` was a mistake) which was subsequently corrected.

Then, between version 10.1.4 and 10.2.3, **no** Grafana release included the `linux/arm/v7` architecture.

That was the situation when I wrote my reply on Discord (above). That was three days ago.

Right now there's a 10.1.6 which *does* include a `linux/arm/v7` architecture. Dockerhub says it was posted 2 hours ago.

I have no idea what's going on with Grafana version numbers. It *may* be that 10.1.x and 10.2.x are different feature streams that are being maintained separately (sort of like InfluxDB 1.8 and InfluxDB 2.x) but please don't take that as gospel.

Anyway, as of now, if you wanted to pin to a version which included a 32-bit architecture, the edit you should make to your `docker-compose.yml` is to make the `image:` clause look like this:

``` yaml
  grafana:
    container_name: grafana
    image: grafana/grafana:10.1.6
    restart: unless-stopped
    …
```

I can, however, assure anyone reading this that the entire problem will just go away if you upgrade your Pi to run a full 64-bit version of Raspberry Pi OS.

gadhvirushiraj, 2023-12-23:
Thanks @Paraphraser, got it working.