SensorsIot/IOTstack #746: Cameras not Working in MotionEye

careyer, 2023-11-26:
Running a Pi3B+ with latest Raspi OS Lite (Bullseye)

with 
`  motioneye:
    image: dontobi/motioneye.rpi:latest
    container_name: "motioneye"
    restart: unless-stopped
    ports:
    - "8765:8765"
    - "8766:8081"
    environment:
    - TZ=${TZ:-Etc/UTC}
    volumes:
    - ./volumes/motioneye/etc_motioneye:/etc/motioneye
    - ./volumes/motioneye/var_lib_motioneye:/var/lib/motioneye
    devices:
    - /dev/video0
    - /dev/video1 
`

neither the mmal camera (CSI camera) nor the connected USB-webcam show up in motioneye.
passing over the devices into docker does not seem to work

Paraphraser, 2023-11-26:
Over on the [IOTstack Discord](https://discord.gg/ZpKHnks) channel, [shiliuhua (aka HS)](https://discord.com/channels/638610460567928832/638610461109256194/1165401965921058886) wrote:

> Hello, I am a newbie, my first time trying to use IOTstack to install related services on raspberry pi4, I met a question about motioneye. I have successfully installed motioneye and Homeassistant through docker compose from IOTstack. When I access motioneye through localhost:8765, I encountered a problem. I want to connect the local usb camera through v4l2, but in the add camera option, v4l2 is displayed below. no camera. May I ask how to solve this problem? Thanks a lot

I [replied](https://discord.com/channels/638610460567928832/638610461109256194/1166159773314596894):

> Hi. I had the same problem a while ago. As far as I'm aware, the vl42 support is broken. I'm not sure whether this is a Raspbian thing or a MotionEye container thing but I have never had any success with "local" cameras in MotionEye since the Bullseye version of Raspbian added LibCamera support.

> My solution was to roll my own mjpg-streamer container. I've made it work with ribbon cameras and a USB camera. In essence, local cameras become network cameras and, as a side effect, that lets you decide whether to run MotionEye and mjpg-streamer on the same or different hosts. There are some performance benefits to running MotionEye and the streamer on the same host - all network traffic is confined to Docker's internal bridged network (a software emulation of a network) so it never hits Ethernet or Wifi, meaning you can stream at 30fps. In my case, MotionEye runs on one host, my cameras on other hosts, and I just stream at 5fps.

> Anyway, following on from your post, I turned my solution into Pull Request #738  and you can read the documentation at [tinyurl.com/3fr6py93](https://tinyurl.com/3fr6py93) to see if sounds like what you need (the original URL on GitHub was horrendously long so I shortened it with TinyURL).

> If you think it will meet your needs then see [this gist](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d#applying-a-single-pull-request) for instructions on how to apply a pull request to your local clone of IOTstack in advance of the PR being approved on IOTstack. The gist also explains how to undo that once the PR is approved.

> If you do decide to try it out, please add any comments, suggestions or other feedback to the pull request page. It will help to know that it works for more than just me.

That traffic was back on October 22/23 but PR #738 is still pending.

I'm mentioning @Slyke here in the hope my very unsubtle hint makes him realise that a lot of electronic water has passed under the digital bridge since the last batch of PRs made its way into production. 😎

Meanwhile, applying PR #738 (the gist link above) should solve your problem.

Hope this helps.

careyer, 2023-11-27:
@Paraphraser Thanks for mentioning your workaround. However, this does not solve the cause of the problem and is a bit like shooting sparrows with cannons. 

Meanwhile I have achieved some success. Likewise the container error message `Failed to open /dev/video0: Permission denied` implies this is not a V4L2 problem but rather a permissions issue. In Bullseye access to `/dev/video0` is restricted. Only the users pi / root and users from the video group do have access to it.
Onbiously the user that runs the docker container does not have access. Since I am new to docker and also not a linux maniac I did not manage to set the correct permission or include the appropriate user to the video group. However I found a dirty workarrond that is much simpler than yours:

`sudo chmod 777 /dev/video0` solves the problem. 
I was able to successfully route a MMAL and  a USB camera to the motioneye container and both work flawlessly.

However file permissions to video decices are reset at each boot.
To overcome this I adeed these lines to `/etc/rc.local`.
```
chmod 777 /dev/video0   // my MMAL/CSI camera
chmod 777 /dev/video1   // my USB Webcam
´´´
Hopefully someone with better linux skills can sort out the permissions issue. Other than that the conatiner works just fine.
Cheers!