SensorsIot/IOTstack #747: Grafana arm/v7 deprecated since 10.1.4

Finnitio, 2023-11-30:
Since 10.1.4 there have not been arm/v7 builds. Only 64bit. builds.

I modiefed docker-compose.yml from
`services:
  grafana:
    container_name: grafana
    image: grafana/grafana`
to
'services:
  grafana:
    container_name: grafana
    image: grafana/grafana:10.1.2'
to work around.

Paraphraser, 2023-12-01:
So, I say this as someone who started using full 64-bit Raspberry Pi OS pretty much as soon as Bullseye came out, and is now using 64-bit Bookworm on both ARM and AMD, I have never once struck a problem that has caused me to consider downgrading to either a full 32-bit system or a mixed-mode 32-bit user side with 64-bit kernel. Containers that only offer a 32-bit implementation run just fine on a full 64-bit system whereas the reverse is not true - as you have just found.

Against that background, I want to ask what combination of circumstances is keeping you running 32-bit? I'd really like to understand any actual 32-bit use cases.

Finnitio, 2023-12-06:
I have a rpi 2b and snce that processor is a arm/v7 it cannot boot arm/v8 kernels aka the 64bit Kernel. Rpi can only use 64 bit from Model 3 and upwards. i am not sure if there are any special OSs that make 64 bit on a rpi possible.

Paraphraser, 2023-12-06:
Ah! A 2B. Now it makes sense.

I think it's probably like everything else in IT - we can resist technological evolution for only so long, after which we have to give in and upgrade.

I *have* managed to get 64-bit and IOTstack running on a Zero2W but the combo of 500MB RAM plus SD makes it quite limiting - only a couple of containers max. The thing has a USB port but I couldn't get it to boot from that.

Right now, I'd say 4GB 4B is the go-to Pi platform.

Finnitio, 2023-12-06:
just wanted to point that out if someone else would run into the same problem :).

Paraphraser, 2023-12-06:
My motivation too. 😎

Finnitio, 2023-12-06:
true but especially for IOT-Sensordata purposes and visualisation it still has enough RAM+CPU Power so it would be shame to throw it away. There seem to be onofficial IOT builds of the newer version in arm/v7,