SensorsIot/IOTstack #754: Integrating ESPHome with My Existing IOTstack Setup

goderz, 2024-02-16:
Hi IOTstack Team,

I've been using IOTstack on my Raspberry Pi and am planning to add ESPHome to my setup for home automation projects. Before I proceed, I wanted to get some insights to ensure a smooth integration.

Here are my main concerns:
- Are there any known compatibility issues with ESPHome in the current IOTstack environment?
- Best practices to avoid port conflicts when adding the ESPHome container?
- Advice on managing ESPHome configurations within Docker?
- Any specific security considerations with ESPHome?

Appreciate any guidance you can provide to help me integrate ESPHome effectively.

Also, I'm quite new to Docker, so I'd appreciate any tips on what to consider when adding more containers in the future.

Thanks,
Paweł.

Paraphraser, 2024-02-17:
Try the following service definition as your starting point:

``` yaml
  esphome:
    container_name: esphome
    image: ghcr.io/esphome/esphome
    restart: unless-stopped
    environment:
      - TZ=${TZ:-Etc/UTC}
      - USERNAME=${ESPHOME_USERNAME:?eg echo ESPHOME_USERNAME=test >>~/IOTstack/.env}
      - PASSWORD=${ESPHOME_PASSWORD:?eg echo ESPHOME_PASSWORD=ChangeMe >>~/IOTstack/.env}
    network_mode: host
    x-ports:
      - "6052:6052"
    x-privileged: true
    volumes:
      - ./volumes/esphome/config:/config
```

This is based upon [ESPHome documentation](https://esphome.io/guides/getting_started_command_line) with adjustments to conform with the norms for IOTstack.

The significant differences are:

1. Rather than map `/etc/localtime` into the container (which, although common, is a bit of a hack), IOTstack passes TZ into the container. You can initialise to your own time-zone by doing this:

	``` console
	$ echo "TZ=$(cat /etc/timezone)" >>~/IOTstack/.env
	```

	This means a single entry in the `.env` file passes TZ to all the containers which support TZ (in essence, those that have the `tzdata` package installed).

2. Rather than provide known defaults for the username and password, this structure of those environment variables forces you to initialise the values yourself (in keeping with best practice):

	``` console
	$ echo "ESPHOME_USERNAME=goderz" >>~/IOTstack/.env
	$ echo "ESPHOME_PASSWORD=my-super-secure-password" >>~/IOTstack/.env
	```

	If you don't, you get a long-winded (and badly laid-out) error message telling you to run those commands.

3. The `x-ports` clause is inactive (any clause starting with `x-` is treated as a comment). The reason for including it is to document the fact the ESPhome uses port 6052. This helps avoid allocating that port to another service.

4. I have disabled the `privileged` flag (by prepending `x-`). I am always suspicious about any service definition I find on the web which includes this flag. It is rarely needed. What it does is let the container break containerisation. Most containers run as root so using this flag gives the container (and its authors) access to your entire host.

	In many cases, this flag seems to be used as a lazy way of providing access to `/dev`, completely ignoring the fact that it also provides unrestricted access to the entire host.

	If a container needs access to specific devices, those should be mapped in, either using a `devices:` clause (providing access to specific devices) or cgroup rules (providing access to a class of devices). For example:

	``` yaml
	devices:
	  - "/dev/ttyACM0:/dev/ttyACM0"
	device_cgroup_rules:
	  - "c 188:* rw"
	```

	I don't use ESPhome so I don't know what it is doing or how it works, so I can't say whether `privileged` is appropriate. I'll leave it to you to sort out whether it is warranted.

5. By convention, IOTstack containers have persistent stores at:

	```
	~/IOTstack/volumes/«containerName»/«path»
	```

	My proposed service definition conforms with this by mapping:

	``` yaml
	volumes:
	  - ./volumes/esphome/config:/config
	```
<hr>
[... 35 lines cut ...]
> Advice on managing ESPHome configurations within Docker?

No. I assume that configurations get written to the **internal** directory:

```
/config
```

which means they will persist to the **external** directory:

```
~/IOTstack/volumes/esphome/config
``` 

but that is something you should verify. Assuming that's the way it works then backup routines will capture the configurations.

> Any specific security considerations with ESPHome?

Other than my reservations about the wisdom (or otherwise) of the `privileged` flag, none.

Hope this helps.

<hr>

If you get it running and can write some documentation for other people to follow then I can help you create a pull request to have this added to IOTstack.

Noschvie, 2024-04-03:
@goderz any feedback ?

goderz, 2024-04-10:
Hey @Noschvie 

Unfortunately, I must point out that I haven't attempted to run ESPHome in the end. I'm doing all of this strictly as a hobby, and I had to focus on something entirely different at the moment. I'll probably return to IoT-related projects later this year. I apologize for the lack of response.

Paraphraser, 2024-04-10:
@goderz that's cool by me

@Noschvie I assume you commented because you're interested in this too. If yes, have you tried it? Does it work? Should it become a Pull Request for being added to IOTstack? Any hints for what should be in documentation to help people get started?

Noschvie, 2024-04-12:
Hello Phill
sorry for the delay. I'm interested, but not a user of ESPHome. Because of an existing integration for HA I think it's not needed to add it to IOTstack.

6ffm70, 2024-07-15:
@Paraphraser thank you for the explanation. I'd love to see ESPhome as part of IOTstack!

PR #773 (2024-09-05): 2024-07-21 Adds ESPHome - master branch - PR 1 of 2
Adds ESPHome container, as requested by #754.

Includes documentation.

Fixes: #754
commit a8900002 (2024-09-05): 2024-07-21 Adds ESPHome - master branch - PR 1 of 2

Adds ESPHome container, as requested by #754.

Includes documentation.

Fixes: #754
closed 2024-09-05 by Slyke