SensorsIot/IOTstack #765: NodeRed outdated Node.js (and npm)

kremik, 2024-04-21:
Hello, I'm trying to install addons in Node Red in this case homekit-service, but the installation fails due to old/unsupported version of Node.js in the nodered image. 

```
2024-04-21T09:11:10.685Z Install : node-red-contrib-homekit-bridged 1.6.1

2024-04-21T09:11:10.815Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict node-red-contrib-homekit-bridged@1.6.1
2024-04-21T09:11:11.240Z [err] npm
2024-04-21T09:11:11.240Z [err]  
2024-04-21T09:11:11.240Z [err] WARN config production Use `--omit=dev` instead.
2024-04-21T09:11:22.307Z [err] npm
2024-04-21T09:11:22.308Z [err]  ERR!
2024-04-21T09:11:22.308Z [err]  code
2024-04-21T09:11:22.308Z [err]  EBADENGINE
2024-04-21T09:11:22.309Z [err] npm 
2024-04-21T09:11:22.309Z [err] ERR! engine Unsupported engine
2024-04-21T09:11:22.309Z [err] npm 
2024-04-21T09:11:22.309Z [err] ERR! engine Not compatible with your version of node/npm: @homebridge/ciao@1.2.0
2024-04-21T09:11:22.309Z [err] npm 
2024-04-21T09:11:22.309Z [err] ERR! notsup Not compatible with your version of node/npm: @homebridge/ciao@1.2.0
2024-04-21T09:11:22.309Z [err] npm 
2024-04-21T09:11:22.309Z [err] ERR! notsup Required: {"node":"^18 || ^20"}
2024-04-21T09:11:22.309Z [err] npm ERR!
2024-04-21T09:11:22.309Z [err]  notsup Actual:   {"npm":"8.19.4","node":"v16.20.2"}
2024-04-21T09:11:22.311Z [err] 
2024-04-21T09:11:22.311Z [err] npm
2024-04-21T09:11:22.311Z [err]  ERR! A complete log of this run can be found in:
2024-04-21T09:11:22.311Z [err] npm 
2024-04-21T09:11:22.311Z [err] ERR!     /data/.npm/_logs/2024-04-21T09_11_11_205Z-debug-0.log
2024-04-21T09:11:22.315Z `rc=1`
```

Paraphraser, 2024-04-21:
Hi, please see [IOTstack Node-RED wiki page](https://sensorsiot.github.io/IOTstack/Containers/Node-RED/#containerVersions).

The IOTstack template for Node-RED defaults to "latest" so we get whatever Node-RED's maintainers think is appropriate. At the moment, that's Node.js 16. If you want 18 then you just change the service definition on your `docker-compose,yml` to look like this:

[image]

To force implementation:

```
$ cd ~/IOTstack
$ docker-compose build --no-cache --pull nodered
$ docker-compose up -d nodered
$ docker system prune -f
```

kremik, 2024-04-23:
@Paraphraser Amazing, failure-proof answer that definitely solved the issue. Thank you!

closed 2024-04-23 by kremik