SensorsIot/IOTstack #806: Support and Document Both docker-compose and docker compose (Plugin) in IOTstack

NossieUK, 2025-06-20:
On Raspberry Pi OS Bookworm, both docker-compose and docker compose are available. With Debian Trixie, only docker compose will be supported.

Requests:

Clearly document the difference between the two tools, and which is default on each OS version.

Add a setup check to show which tool is installed and active.

Would this be a big change or something that only impacts startup?

NossieUK, 2025-06-21:
I am so sorry Phil - I appear to have wasted your time without doing my own research

I upgraded to Trixie as a test on one of my systems and the iotstack stopped working - I immediately assumed it was  a new thing in Trixie - however I never tried to reinstall the iotstack script 

[image]

Although I have managed to break the virtual environment trying to test - so trixie must have broke something upgrading from bookworm \o/

apologies again though - thanks.

Paraphraser, 2025-06-22:
No, **not** wasted. I always learn a lot from questions like this. I now know about that dependency in the 03 script. I also know the Python side of things is going to involve a bit of work.

Paraphraser, 2025-07-04:
> so trixie must have broke something upgrading from bookworm

That's a definite possibility. My primary IoT systems are Raspberry Pis which are all currently running Bullseye. When I first started on the Pi path, 32-bit Buster was all the rage. When I upgraded to Bullseye, I rebuilt each system from scratch. I was following advice which said upgrading in situ is possible but not recommended.

Recently I've started to think about upgrading those IoT systems to Bookworm. I've never run anything other than Bookworm on Proxmox-VE so I'm quite happy with it and, other than finding the time, there's no reason not to upgrade. I took a test Pi which was running Bullseye and did an upgrade in situ to see what would happen. It *looks* like it is working but I keep finding things that aren't as I expect, beginning with NetworkManager not being there.

Thus I'm adding my voice to the people who say "possible but not recommended" and I'll take the time to rebuild my own systems from scratch.

> thanks again .. strangely I could only get it to work after running this

Well, that makes no sense at all. This is from the Trixie system I spun up on Proxmox-VE the other day:

[image]

Putting both of these things together, I'm starting to wonder if your system has a much longer history and if, perhaps, you had some ancient versions of docker and/or docker-compose installed at one point.

docker-compose has been through three iterations (that I'm aware of). It started as a compiled binary, then became a Python script, then back to a compiled binary again.

On anything up to Bookworm (and possibly Trixie) if you do something like:

```
$ sudo apt install docker docker-compose
```

you get the 1st-gen docker-compose and that forces a permanent downgrade of docker, after which `apt` won't update either docker or docker-compose ever again. Unfortunately, examples installing docker-compose using `apt` are fairly common and they are wrong.

Now, PiBuilder has some scripts to deal with this. If any containers are running, you need to stop them. Then:

```
$ ~/PiBuilder/boot/scripts/helpers/uninstall_docker.sh
$ ~/PiBuilder/boot/scripts/helpers/uninstall_docker-compose.sh
$ sudo reboot
$ ~/PiBuilder/boot/scripts/04_setup.sh
```

The first script is fairly straightforward. It just does what the name implies.

The second script goes after all possible docker-compose iterations (compiled, Python, compiled) and clobbers them.

The reboot gets rid of dross that prevents a clean re-install.

The 04 script does a proper install of docker using the "convenience script" which brings in docker-compose-plugin, after which the symlink is set up.

The 04 script does a logout at the end (because it alters group membership) so you need to login again.

I've just run through that sequence on my Trixie system and it works.

<hr>

Another possibility, I suppose, is your PATH is set in a way which I haven't considered. Have you tried:

```
$ which -a docker-compose
```

If you get more than one hit then that might explain *some* of what you are seeing. I would, however, not expect your approach of creating a bash script would have cured the problem because it would still be in the same relative position in your PATH.